The-OpenROAD-Project/OpenROAD #4: Build problems on Ubuntu 18.04

byzantic, 2019-11-27:
Hi I get a couple of problems when trying to build on Ubuntu 18.04

On Ubuntu, the TCL headers are in `/usr/include/tcl/tcl.h`, and compilations claim they can't find `tcl.h`. Which is odd, because the output of CMAKE seems to imply that it has found it correctly.

Also during a link step, involving pthreads, there was a complaint about a missing DSO file.

Keep up the good work!

rovinski, 2019-12-03:
Hi @byzantic,

Thanks for the concern. We are currently in a development-heavy state and focused only on supporting CentOS 7. We may consider compatibility with other OSs at some point in the future, but it is not on our roadmap. We recommend using Docker to build OpenROAD on non-CentOS 7 systems.

I'll leave this issue open for the time being - if the issue queue becomes too messy, it may be closed until we can consider it in the future.

byzantic, 2019-12-03:
Sure, I understand - You all look as though you are very busy!

tamimcse, 2019-12-05:
I also had same issue on Ubuntu. I used following command to include the TCL include path. 

export CPLUS_INCLUDE_PATH=$CPLUS_INCLUDE_PATH:/usr/include/tcl

Then added pthread to Cmake as following:
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index c0d5061..f94313a 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -187,6 +187,7 @@ target_link_libraries(openroad
   lefin
   lefout
   zutil
+  pthread
   ${TCL_LIB}
   )

Now it compiles on Ubuntu.

jjcherry56, 2019-12-11:
I was able to build 05e4818 on ubuntu 18.04. Hurry now, this offer may not last!

closed 2019-12-11 by jjcherry56

byzantic, 2019-12-12:
Err .. I must have bobbled it somewhere when checking out the develop branch, as I get an error about there being no CMakeLists.txt in Fastroute.

But in any case, I'm obviously jumping the gun here .. it's far too early to start thinking about such details! Keep up the good work!

rovinski, 2019-12-12:
I've run into that issue sometimes. Usually it's caused by not running `git submodule init; git submodule update --recursive`. Git doesn't always seem to recognize the multiple levels of submodules, so you may need to `cd` into `FastRoute` and then update the submodules.