The-OpenROAD-Project/OpenROAD #44: Running openroad in a different directory

Colin-Holehouse, 2019-12-18:
If I run openroad within the directory where the executable was built, then the tool starts up successfully. However if I move up one directory, then the tool exits as shown in the transcript below

````
login8.euhpc > openroad
% exit
login8.euhpc > cd ..
login8.euhpc > openroad
Error: could not find FluteLUT files POWV9.dat and POST9.dat.
````

Seems like the mechanism that the tool uses to find these files should be improved.

Colin-Holehouse, 2019-12-18:
If I run the tool using the absolute path to the executable, rather than letting it resolve the location of the executable via the PATH environment variable, then the tool starts up corectly

rovinski, 2019-12-20:
The `dat` files are generally an annoying part of flute. I think the current workaround is to have symlinks to the `dat` files in the directory you are running in. We need to look into this at some point.

Colin-Holehouse, 2019-12-20:
Using symlinks is a horrible hack. There is code that looks for these files in relation to where the executable is found - it should just account for being invoked through the PATH mechanism.

I plan to keep using an absolute path to call the tool until this gets resolved.

jjcherry56, 2019-12-20:
It was vastly improved; by me.
Before openroad every tool was copying the flute files and leaving turds all over the file system.
At least now it knows to look around for the files relative to the executable.
You should look at what a mess the alpha-flow is because of it.
Until you spoke up I seemed to be the only one that thought this was a problem.
The braziliian team has had the task to fix flute so these files are part of the executable for months based on my request.
I am not going to waste time chasing path variables in the meantime.

rovinski, 2019-12-21:
@Colin-Holehouse there are a ton of hacks floating around right now, this one is comparatively very minor. I'm hoping we can focus on cleaning things up after the January demonstration.

aolofsson, 2020-01-24:
So is the workaround for now to call openroad with a full path?

/usr/local/bin//openroad

aolofsson, 2020-01-24:
I second James' request! This feels wrong that the file be embedded in the executable.

eder-matheus, 2020-01-31:
Please, check the commit 70552fd40340d5e403ba298499cf25fccc5e7d1a. It embeds the LUTs in the binary, allowing it to be executed in any place.

jjcherry56, 2020-01-31:
please remove the initFlute function, header and .cc file and just call the
init function
in OpenRoad::init. There is no reason to leave all of that behind.

jjcherry56, 2020-01-31:
also remove the .DAT files in /etc (and /etc/ because it will be empty) and
remove the install target the copies them in the top level cmake file

eder-matheus, 2020-01-31:
@jjcherry56 done. Please, check the last commits of "[fix/flute_lut](https://github.com/The-OpenROAD-Project/OpenROAD/tree/fix/flute_lut)" branch

jjcherry56, 2020-02-01:
I am sure it is fine. It passes the regressions so any other clean up can
be done after it is merged.
Go ahead and do it. Thanks for following through on this.

jjcherry56, 2020-02-01:
how could you not notice the compiler warning this introduced?

/Users/cherry/sta/openroad/src/OpenRoad.cc:96:21: warning: unused parameter
'prog_arg'
      [-Wunused-parameter]
               const char *prog_arg)
                           ^
1 warning generated.

jjcherry56, 2020-02-01:
Once you merge a branch you should delete it.
You guys seem to keep branches around for weeks. Uncool.

eder-matheus, 2020-02-02:
Branch deleted; warning fixed in the last commit of "openroad" branch.

jjcherry56, 2020-02-02:
Thank you.
Here's another one

/Users/cherry/sta/openroad/src/TritonCTS/src/DbWrapper.cpp:247:40: warning:
comparison of
      two values with different enumeration types ('odb::dbIoType::Value'
and
      'Ath__termDirection') [-Wenum-compare]
                if (iterm->getIoType() != INPUT) {
                    ~~~~~~~~~~~~~~~~~~ ^  ~~~~~
1 warning generated.

mpfogaca, 2020-02-02:
@jjcherry56 I will fix TritonCTS warning.

closed 2020-02-04 by jjcherry56