The-OpenROAD-Project/OpenROAD #83: read_def does not support engineering notation for dimensions

teadotjay, 2020-01-09:
I'm on site for the integration exercise and investigating using the router in a mixed flow with a third-party placer. The placer uses engineering notation (e.g. 1.89e+06 for 1890000) in all the geometry dimensions, and this is causing syntax errors in openroad. If I replace these with the numeric notation, it reads in successfully.

test.def:
```def
VERSION 5.8 ;
DIVIDERCHAR "/" ;
BUSBITCHARS "[]" ;
DESIGN test ;
UNITS DISTANCE MICRONS 1000 ;

DIEAREA ( 0 0 ) ( 1.89e+06 1.5e+06 ) ;

END DESIGN
```

test2.def:
```def
VERSION 5.8 ;
DIVIDERCHAR "/" ;
BUSBITCHARS "[]" ;
DESIGN test2 ;
UNITS DISTANCE MICRONS 1000 ;

DIEAREA ( 0 0 ) ( 1890000 1500000 ) ;

END DESIGN
```

Example:
```tcl
% read_def test.def
Notice 0: 
Reading DEF file: test.def
Notice 0: Design: test
Notice 0: error: DEF:7: syntax error, reading 1.89e+06
```

```tcl
% read_def test2.def
Notice 0: 
Reading DEF file: test2.def
Notice 0: Design: test2
Notice 0: Finished DEF file: test2.def
```

jjcherry56, 2020-01-10:
That isn't legal DEF. Read the last sentence below.

Die Area
[DIEAREA pt pt [pt] ... ;]
If two points are defined, specifies two corners of the bounding rectangle for the design. If
more than two points are defined, specifies the points of a polygon that forms the die area.
The edges of the polygon must be parallel to the x or y axis (45-degree shapes are not
allowed), and the last point is connected to the first point. All points are integers, specified as
DEF database units.

closed 2020-01-10 by jjcherry56

teadotjay, 2020-01-18:
It seems you are correct. I will file a bug report against the tool generating this notation.