The-OpenROAD-Project/OpenROAD #85: fastroute won't run if fewer than 3 metal layers

teadotjay, 2020-01-09:
My technology has only two routing layers. To get fastroute to run, I had to create a third dummy metal layer and the associated tracks, then set -max_routing_layer 2.

```tcl
% fastroute -max_routing_layer 2
WARNING: Default output guide name: out.guide
WARNING: Default capacity adjustment: 0.0 (no adjustment)
WARNING: Default minimum layer: 1
Initializing grid...
[ERROR] Layer3 not found! Exiting...
```

rovinski, 2020-01-14:
Hi @teadotjay,

Is this a real technology? Any real metal stack should have at least 3 layers. The problem with having fewer than 3 layers is that it breaks modern routing methodologies. Metal1 is used for creating standard cell rails and therefore cannot be used route except to adjacent cells in the same row. Metals 2 and 3 are therefore used for horizontal and vertical routing. If there is no Metal3, then the tool cannot route both horizontally and vertically, and would therefore break the algorithm.

If this is an analog technology, keep in mind that FastRoute is a global router intended for digital processes. You may need to use a different global router if this is the case.

teadotjay, 2020-01-14:
Hi, Austin,

Yes, this is a real digital technology, but it's a bit unconventional. The
global signal routing is on M1-M2 with higher level metals reserved for
ground planes, power distribution, and the cell electronics, as the active
layer is actually between metal layers rather than beneath as in CMOS. This
works in qrouter as well as NanoRoute (Cadence Innovus).

As a workaround, I managed to get the global router to route in 2 metal
layers using the -max_routing_layer (or similarly named) command switch and
a modified LEF containing a dummy third metal, but then I have to use the
2-metal LEF for TritonRoute or it tries to jump up to the third layer and
short with a ground plane. Which means I have to create a dummy M3 track in
the DEF file, then subsequently delete it for TritonRoute.

Thanks,
T.J.

rovinski, 2020-01-14:
@eder-matheus is it possible to remove this restriction?

@teadotjay it would greatly help if you could send us your inputs for FastRoute, if not covered by NDA. If you could create a fake LEF that exhibits the issue, that would also greatly help. We can't guarantee timely responses unless we have the inputs to recreate the issue.