The-OpenROAD-Project/OpenROAD #96: Build fails in Ubuntu

aolofsson, 2020-01-22:
For collaboration purposes, I do think it should be possible to build OpenROAD from source. Log file based on following README file below. I am guessing dependancy issues, but there are no extract requirementsgiven, so not sure how to reproduce build.

-- OpenROAD version: 1.0.1
-- System name: Linux
-- Compiler: GNU 7.4.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- CXX_FLAGS: 
-- Install prefix: /usr/local
-- TCL lib: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- STA version: 2.0.17
-- System name: Linux
-- Compiler: GNU 7.4.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- STA CXX_FLAGS: -Wall;-Wextra;-pedantic;-Wcast-qual;-Wredundant-decls;-Wformat-security
-- Install prefix: /usr/local
-- CUDD library: not found
-- SSTA: 0
-- TCL lib: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- STA executable: /home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/app/sta
-- ioPlacer
-- /home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/ioPlacer
INFO: RELEASE BUILD
-- Boost version: 1.65.1
-- TCL lib: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found GLPK: /usr/lib/x86_64-linux-gnu/libglpk.so (Required is at least version "4.33") 
-- Could NOT find ILOG (missing: ILOG_CPLEX_LIBRARY ILOG_CPLEX_INCLUDE_DIR) 
-- Could NOT find COIN (missing: COIN_INCLUDE_DIR COIN_CBC_LIBRARY COIN_CBC_SOLVER_LIBRARY COIN_CGL_LIBRARY COIN_CLP_LIBRARY COIN_COIN_UTILS_LIBRARY COIN_OSI_LIBRARY COIN_OSI_CBC_LIBRARY COIN_OSI_CLP_LIBRARY) 
-- Could NOT find SOPLEX (missing: SOPLEX_LIBRARY SOPLEX_INCLUDE_DIR) 
CMake Error at src/CMakeLists.txt:132 (add_subdirectory):
  The source directory

    /home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/FastRoute

  does not contain a CMakeLists.txt file.

CMake Error at src/CMakeLists.txt:134 (add_subdirectory):
  The source directory

    /home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/TritonMacroPlace

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
See also "/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/build/CMakeFiles/CMakeOutput.log".

maliberty, 2020-01-22:
Are you doing this on the 'master' branch?  On line 132 I just see a ')' on master (https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/CMakeLists.txt).

rovinski, 2020-01-22:
This is likely related to the submodules not being cloned. Either the OpenROAD repository has to be freshly cloned with `git clone --recursive` or for each submodule (e.g. each DIRECTORY that says it has no `CMakeLists.txt`) do
```
cd $DIRECTORY
git submodule init
git submodule update
```
In theory `git submodule update --init --recursive` should work, but I think sometimes it doesn't for an unknown reason.

aolofsson, 2020-01-22:
Got further with alpha 2.1.

```bash
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git --branch alpha2.1
mkdir -p OpenROAD/build
cd OpenROAD/build
cmake ..
cmake ..
make
```

Two potential issues:

1.) I had to enter "cmake .." twice to be able to make, caching problem?

2.) Failing on "tcl.h", likely due to ubuntu vs centos difference?

Ubuntu path should be : /usr/include/tcl/tcl.h

Current path detected by cmake is: /usr/include/tcl

Need to look for type of distro in cmake/file? OpenROAD/cmake/FindTcl.cmake
And set accordingly.

==========================================================
[ 62%] Building CXX object src/dbSta/CMakeFiles/dbSta.dir/dbSta.cc.o
cd /home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/build/src/dbSta && /usr/bin/c++  -DOPENROAD_BUILD -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/include -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/util -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/liberty -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/network -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/graph -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/sdc -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/parasitics -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/dcalc -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/sdf -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/search -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenSTA/app -I/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/OpenDB/include  -O3 -DNDEBUG   -std=c++11 -o CMakeFiles/dbSta.dir/dbSta.cc.o -c /home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/dbSta/dbSta.cc
/home/aolofsson/work/darpa/IDEA/ucsd-openroad/OpenROAD/src/dbSta/dbSta.cc:17:10: fatal error: tcl.h: No such file or directory
 #include <tcl.h>

ckqee, 2020-01-22:
```
OpenROAD/build/src/dbSta/dbSta_wrap.cc:195:10: fatal error: tcl.h: No such file or directory
 #include <tcl.h>
          ^~~~~~~
compilation terminated.
src/dbSta/CMakeFiles/dbSta.dir/build.make:146: recipe for target 'src/dbSta/CMakeFiles/dbSta.dir/dbSta_wrap.cc.o' failed
make[2]: *** [src/dbSta/CMakeFiles/dbSta.dir/dbSta_wrap.cc.o] Error 1
make[2]: Leaving directory 'OpenROAD/build'
CMakeFiles/Makefile2:1184: recipe for target 'src/dbSta/CMakeFiles/dbSta.dir/all' failed
make[1]: *** [src/dbSta/CMakeFiles/dbSta.dir/all] Error 2
make[1]: Leaving directory 'OpenROAD/build'
Makefile:132: recipe for target 'all' failed
make: *** [all] Error 2

The file OpenROAD/build/src/dbSta/dbSta_wrap.cc is autogenerated, a quick peek at it shows the following:

/* ----------------------------------------------------------------------------
 * This file was automatically generated by SWIG (http://www.swig.org).
 * Version 3.0.12
 *
 * This file is not intended to be easily readable and contains a number of
 * coding conventions designed to improve portability and efficiency. Do not make
 * changes to this file unless you know what you are doing--modify the SWIG
 * interface file instead.
 * ----------------------------------------------------------------------------- */
```

So this has to do with how SWIG gets invoked.

CK

ckqee, 2020-01-22:
Doing the following

`git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git`
`mkdir -p OpenROAD/build`
`cd !$`
`cmake ..`
`make -j4`

We get:

`[ 14%] Building CXX object src/dbSta/CMakeFiles/dbSta.dir/dbSta_wrap.cc.o
cd OpenROAD/build/src/dbSta && /usr/bin/c++   -IOpenROAD/include -IOpenROAD/src/OpenSTA/util -IOpenROAD/src/OpenSTA/liberty -IOpenROAD/src/OpenSTA/network -IOpenROAD/src/OpenSTA/graph -IOpenROAD/src/OpenSTA/sdc -IOpenROAD/src/OpenSTA/parasitics -IOpenROAD/src/OpenSTA/dcalc -IOpenROAD/src/OpenSTA/sdf -IOpenROAD/src/OpenSTA/search -IOpenROAD/src/OpenSTA/app -IOpenROAD/src/OpenDB/include -isystem /usr/include/tcl  -O3 -DNDEBUG   -std=c++11 -o CMakeFiles/dbSta.dir/dbSta_wrap.cc.o -c OpenROAD/build/src/dbSta/dbSta_wrap.cc
[ 14%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiMacro.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiMacro.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiMacro.cpp
[ 15%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiMisc.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiMisc.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiMisc.cpp
[ 15%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiNonDefault.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiNonDefault.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiNonDefault.cpp
[ 15%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiProp.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiProp.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiProp.cpp
[ 15%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiPropType.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiPropType.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiPropType.cpp
[ 15%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiTBExt.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiTBExt.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiTBExt.cpp
[ 16%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiUnits.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiUnits.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiUnits.cpp
[ 16%] Building CXX object src/OpenDB/src/lef56/CMakeFiles/opendblef.dir/lef/lefiVia.cpp.o
cd OpenROAD/build/src/OpenDB/src/lef56 && /usr/bin/c++   -IOpenROAD/src/OpenDB/include/lef56 -IOpenROAD/src/OpenDB/src/lef56/lef -IOpenROAD/build/src/OpenDB/src/lef56 -IOpenROAD/src/OpenDB/src/lef56/. -isystem /usr/include/tcl  -O3 -DNDEBUG -fPIC   -std=c++11 -o CMakeFiles/opendblef.dir/lef/lefiVia.cpp.o -c OpenROAD/src/OpenDB/src/lef56/lef/lefiVia.cpp
[ 16%] Building CXX object src/resizer/src/CMakeFiles/resizer.dir/SteinerTree.cc.o
cd OpenROAD/build/src/resizer/src && /usr/bin/c++   -IOpenROAD/src/resizer/include -IOpenROAD/include -IOpenROAD/src/OpenSTA/util -IOpenROAD/src/OpenSTA/liberty -IOpenROAD/src/OpenSTA/network -IOpenROAD/src/OpenSTA/graph -IOpenROAD/src/OpenSTA/sdc -IOpenROAD/src/OpenSTA/parasitics -IOpenROAD/src/OpenSTA/dcalc -IOpenROAD/src/OpenSTA/search -IOpenROAD/src/OpenSTA/app -IOpenROAD/src/OpenDB/include -IOpenROAD/src/flute3 -isystem /usr/include/tcl  -O3 -DNDEBUG   -std=c++11 -o CMakeFiles/resizer.dir/SteinerTree.cc.o -c OpenROAD/src/resizer/src/SteinerTree.cc`

So we need the updated CMakefile.txt to be backported to the alpha release.
CK

ckqee, 2020-01-22:
Forgot to mention that I am using the following Ubuntu version:

`Linux machine 4.15.0-72-generic #81-Ubuntu SMP Tue Nov 26 12:20:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

jjcherry56, 2020-01-23:
Unfortunately there is quite a bit of disinformation in the replies to this issue.
Active development happens in the openroad branch.
Unfortunately, that means that ports that are not the same as the continuous integration
machine can be inadvertently broken.
The master branch has not been updated for a long time (too long).
The README.md file there contains build instructions that work.
I build openroad regularly on an aws instance running ubuntu to find portability issues (there are always new ones).
Because of your difficulties I updated the master branch to point a recent commit
on the openroad branch. You should be able to build master on ubuntu now.

aolofsson, 2020-01-24:
Thank you James!! Build now works for me in Ubuntu 18.04!

closed 2020-01-24 by aolofsson