The-OpenROAD-Project/OpenROAD #102: read_lef history.tcl error

aolofsson, 2020-01-24:
Testing main branch build of openroad with test/gcd_flow1.tcl.
(other than that, got through def)

% read_lef NangateOpenCellLibrary.lef
Error: history.tcl, 330 invoked "return" outside of a proc.
Notice 0: Reading LEF file:  NangateOpenCellLibrary.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  NangateOpenCellLibrary.lef

jjcherry56, 2020-01-25:
I see those from time to time but I don't know why they happen. 
history.tcl is part of the tcl install.
The are innocuous at least.

jjcherry56, 2020-01-28:
I fixed this in the openroad branch with commit 001cab3

closed 2020-01-31 by jjcherry56