The-OpenROAD-Project/OpenROAD #103: Build Error in OpenRCX

abdelrahmanhosny, 2020-01-26:
I'm using SUSE Linux:
```
NAME="SLES"
VERSION="15-SP1"
VERSION_ID="15.1"
PRETTY_NAME="SUSE Linux Enterprise Server 15 SP1"
ID="sles"
ID_LIKE="suse"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:suse:sles:15:sp1"
```

Here is the CMake output:
```
-- OpenROAD version: 1.0.1
-- OpenROAD git sha: bcc20ec161950770c1b013476ff11ed852434ad0
-- System name: Linux
-- Compiler: Clang 7.0.1
-- Build type: RELEASE
-- Build CXX_FLAGS: 
-- CXX_FLAGS: 
-- Install prefix: /usr/local
-- TCL library: /usr/lib64/libtcl8.6.so
-- TCL header: /usr/include/tcl.h
-- TCL library: /usr/lib64/libtcl8.6.so
-- TCL header: /usr/include/tcl.h
-- leflib_home: /home/ec2-user/OpenROAD/src/OpenDB/src/lef
-- STA version: 2.0.17
-- STA git sha: 25bf7b7ac87cd90d82a8e4ee18cf0a59d6e6530c
-- System name: Linux
-- Compiler: Clang 7.0.1
-- Build type: RELEASE
-- Build CXX_FLAGS: 
-- STA CXX_FLAGS: -Wall;-Wextra;-pedantic;-Wcast-qual;-Wredundant-decls;-Wformat-security;-Wno-deprecated-register
-- Install prefix: /usr/local
-- TCL library: /usr/lib64/libtcl8.6.so
-- TCL header: /usr/include/tcl.h
-- CUDD library: not found
-- SSTA: 0
-- STA executable: /home/ec2-user/OpenROAD/src/OpenSTA/app/sta
-- TCL lib: /usr/lib64/libtcl8.6.so
-- TCL header: /usr/include/tcl.h
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Could NOT find Ghostscript (missing: GHOSTSCRIPT_EXECUTABLE) 
-- Could NOT find GLPK (missing: GLPK_LIBRARY GLPK_INCLUDE_DIR GLPK_PROPER_VERSION_FOUND) (Required is at least version "4.33")
-- Could NOT find ILOG (missing: ILOG_CPLEX_LIBRARY ILOG_CPLEX_INCLUDE_DIR) 
-- Could NOT find COIN (missing: COIN_INCLUDE_DIR COIN_CBC_LIBRARY COIN_CBC_SOLVER_LIBRARY COIN_CGL_LIBRARY COIN_CLP_LIBRARY COIN_COIN_UTILS_LIBRARY COIN_OSI_LIBRARY COIN_OSI_CBC_LIBRARY COIN_OSI_CLP_LIBRARY) 
-- Could NOT find SOPLEX (missing: SOPLEX_LIBRARY SOPLEX_INCLUDE_DIR) 
-- TCL library: /usr/lib64/libtcl8.6.so
-- TCL header: /usr/include/tcl.h
-- ABKCommon_home: /home/ec2-user/OpenROAD/src/TritonMacroPlace/module/ABKCommon
-- ParquetFP_home: /home/ec2-user/OpenROAD/src/TritonMacroPlace/module/ParquetFP
-- TCL library: /usr/lib64/libtcl8.6.so
-- TCL header: /usr/include/tcl.h
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ec2-user/OpenROAD/build
```

I get the following error
```
/home/ec2-user/OpenROAD/src/OpenRCX/src/ext.cpp:45:36: error: cannot define or redeclare '_module' here because namespace 'OpenRCX' does not enclose namespace
      'ZTechModule<OpenRCX::Ext>'
const char* odb::ZTechModule<Ext>::_module = nullptr;
```

I'm attaching the full `make` output.
[make.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/4113500/make.log)

maliberty, 2020-01-27:
Please try again with the latest head of openroad.

abdelrahmanhosny, 2020-01-27:
The head of `openroad` branch gives this error:
```
[  0%] Building CXX object src/CMakeFiles/openroad_swig.dir/OpenRoadTCL_wrap.cxx.o
cd /home/ec2-user/OpenROAD/build/src && /usr/bin/clang++  -DOPENROAD_BUILD  -O3 -DNDEBUG   -std=c++11 -o CMakeFiles/openroad_swig.dir/OpenRoadTCL_wrap.cxx.o -c /home/ec2-user/OpenROAD/build/src/OpenRoadTCL_wrap.cxx
/home/ec2-user/OpenROAD/build/src/OpenRoadTCL_wrap.cxx:1700:10: fatal error: 'opendb/db.h' file not found
#include "opendb/db.h"
         ^~~~~~~~~~~~~
1 error generated.
make[2]: *** [src/CMakeFiles/openroad_swig.dir/build.make:72: src/CMakeFiles/openroad_swig.dir/OpenRoadTCL_wrap.cxx.o] Error 1
make[2]: Leaving directory '/home/ec2-user/OpenROAD/build'
make[1]: *** [CMakeFiles/Makefile2:135: src/CMakeFiles/openroad_swig.dir/all] Error 2
```

abdelrahmanhosny, 2020-01-28:
Issue resolved.

closed 2020-01-28 by abdelrahmanhosny