The-OpenROAD-Project/OpenROAD #108: Failed git clone

RTimothyEdwards, 2020-02-07:
I tried today to do a git clone of the repo according to the instructions, but part way through I got the error:

error: Server does not allow request for unadvertised object 10e8a5378637808ddaa1c313158e39d587c9bc38
Fetched in submodule path 'src/opendp', but it did not contain 10e8a5378637808ddaa1c313158e39d587c9bc38.  Direct fetching of that commit failed.

Any ideas?

rovinski, 2020-02-07:
I see this issue too. It appears that the submodule `opendp` is set to track from branch `openroad` which has just been deleted. The-OpenROAD-Project/OpenDP@10e8a53 still exists but git throws an error because it is not part of the branch it is expecting.

Can someone update the opendp submodule to track the correct branch? @maliberty @jjcherry56 ?

mithro, 2020-02-07:
This is a GitHub issue that doesn't allow you to clone commits that are not included in a public branch.

A quick solution is to push a branch to the **OpenDP repo** pointing to 10e8a53 commit.

maliberty, 2020-02-07:
Opendp was a submodule but now is part of the OpenROAD repo.  Git doesn't seem to be very smart about this case but I found just removing src/opendp and running git pull works (assuming you don't have any changes in opendp you want to keep).

jjcherry56, 2020-02-07:
Matt, that doesn't work if you can't clone the repo.
Even removing the branch in the .gitmodules failled.
I just moved master to the lastest openroad commit and it works now.

closed 2020-02-07 by jjcherry56