The-OpenROAD-Project/OpenROAD #109: Unmentioned and uninstallable dependencies

RTimothyEdwards, 2020-02-07:
Thanks for the prompt response on the git clone problem.

I am still attempting to compile from source.  On doing cmake, I get complaints about not finding these dependencies, which are not mentioned in the dependency list in the README file:

GLPK, ILOG, COIN, SOPLEX

For the first one, I tried "apt install libglpk-dev" but got back:
   libglpk-dev : Depends: libgmp-dev but it is not going to be installed
                       Depends: libsuitesparse-dev but it is not going to be installed

The others do not appear to be things that can be obtained using yum or apt.
Thoughts?
---Tim

rovinski, 2020-02-07:
All four dependencies are optional dependencies of [lemon](https://lemon.cs.elte.hu/trac/lemon), a library used for TritonCTS. You can find more details in the [INSTALL](https://github.com/The-OpenROAD-Project/OpenROAD/blob/26437d70f094abf564317c25803fd93a80f6dcc0/src/TritonCTS/src/third_party/lemon/INSTALL) file.

They should not be required to install in order to run TritonCTS. You can take a look at the [Dockerfile](https://github.com/The-OpenROAD-Project/OpenROAD/blob/26437d70f094abf564317c25803fd93a80f6dcc0/Dockerfile) to view look at all of the dependencies we install.

mpfogaca, 2020-02-07:
cmake just let you know that those optional dependencies have not being found, please consider it as a warning, not an error.  You should see the message "-- Build files have been written to:...." at the end of cmake execution, confirming that cmake finished its execution without errors.

RTimothyEdwards, 2020-02-07:
Okay, got it.  Is there any way to make the message from cmake less dire-sounding?

mpfogaca, 2020-02-07:
Hi, @RTimothyEdwards. I'll remove that checks since we are not using those packages anyway. Sorry for the inconvenience.

RTimothyEdwards, 2020-02-07:
Not a problem.  Thanks for the explanations and patience.  I have built bits and pieces of OpenROAD back during the early development stages, so I had come to expect compile problems, but I have to say that was an amazingly smooth and clean compile.  I'm impressed with how far you've taken it.  Now it's time to learn how to use it properly. . .

closed 2020-02-07 by RTimothyEdwards