The-OpenROAD-Project/OpenROAD #114: initialize floorplan overflow

kareefardi, 2020-02-11:
Hello,

I was trying initialize floorplan and specifying `-core_area` and `-die_area`. Leading to a floorplan with 0 rows and invalid diearea `DIEAREA ( -2147483648 -2147483648 ) ( 0 0 ) ;`. I narrowed it down to possible integer overflow while carrying out operations in `metersToMfgGrid` as follows: https://github.com/The-OpenROAD-Project/OpenROAD/blob/eec88b45b938e8cff79d7701311546d9f7a49523/src/InitFloorplan.cc#L603

Thanks

jjcherry56, 2020-02-16:
what were the command args? ie, where is the testcase?

jjcherry56, 2020-02-21:
I suspect the issue is that no liberty file had been read before init_floorplan so the units were not defined and so your 100 microns was interpreted as 100 meters. I only know this because I tripped across the same issue. I added a check to make sure there were valid units defined before init_floorplan.

closed 2020-02-21 by jjcherry56

kareefardi, 2020-02-21:
Works fine after reading a liberty file, didn't think of that as the liberty has no units for distance