The-OpenROAD-Project/OpenROAD #126: opendp overlapping cells

kareefardi, 2020-02-24:
Hello,
Opendp places cells over tapcells in the middle of the floorplan. It seems that providing the same input parameters (def and lef) to opendp that a legacy version of opendp doesn't produce the same wrong behavior (specifically this version https://github.com/sanggido/OpenDP).
Thank you

jjcherry56, 2020-02-24:
no test case, no response

kareefardi, 2020-02-24:
@jjcherry56 Sorry should have mentioned this. Would you please check with @tspyrou for a private repo for the test case

jjcherry56, 2020-02-24:
roger.  happy to look at it if there is a test case

jjcherry56, 2020-02-24:
I have the testcase now. Can you point me to a specific instance that is problematic? I don't have any graphics to make it obvious.

kareefardi, 2020-02-24:
Hello, these are the cells that are reported overlapping from opendp. Is that what you are asking for? I can provide screenshots if needed
```
overlap_check ==> FAIL ( cell PHY_72 overlaps _757_ )  ( 83860, 20360 )
overlap_check ==> FAIL ( cell PHY_73 overlaps _582_ )  ( 43840, 23760 )
overlap_check ==> FAIL ( cell PHY_76 overlaps _576_ )  ( 83860, 33960 )
overlap_check ==> FAIL ( cell PHY_83 overlaps _490_ )  ( 43840, 57760 )
overlap_check ==> FAIL ( cell PHY_92 overlaps _457_ )  ( 83860, 88360 )
overlap_check ==> FAIL ( cell PHY_93 overlaps _733_ )  ( 43840, 91760 )
overlap_check ==> FAIL ( cell PHY_95 overlaps _476_ )  ( 43840, 98560 )
overlap_check ==> FAIL ( cell PHY_96 overlaps _467_ )  ( 83860, 101960 )
```

jjcherry56, 2020-02-25:
That's exactly what I'm looking for.

jjcherry56, 2020-02-25:
How did you get that report?
When I run the stand alone opendp it aborts because there are no
power/ground nets defined.
I can only get that result if I remove the exit and recompile.

jjcherry56, 2020-02-25:
fixed by f510c1a opendp issue 126

closed 2020-02-25 by jjcherry56

kareefardi, 2020-02-25:
@jjcherry56 Thanks for the effort. I tried it and they don't overlap now. However, The gap between the tap cell and the neighboring cells doesn't seem to respect the min cell/site width (0.46 in this case). I attached a screenshot describing that. Same test case as before. This specific cell is "PHY_92" among others like PYH_89 and PHY_68. If you would like, I can open another issue.
[image]

jjcherry56, 2020-02-25:
Excuse our ignorance, but where is the gap specified? I think I know how to
address it, but I can't if I don't know what the rules are. I'm asking on
this end also

kareefardi, 2020-02-25:
If by gap you are referring to the 0.46, then that's the site width defined in the lef.

But on further inspection of the test case, I noticed that tapcells are misaligned. For instance, in this row:
```
ROW ROW_0 <site> 3360 3360 FS DO 248 BY 1 STEP 460 0 ;
``` 
There is this tapcell:
```
PHY_66 <tap cell> + FIXED ( 43700 3360 ) FS ;
```
Since 43700-3360 is not a multiple of 460 (the site width), I think this misalignment will cause issues with having weird gaps between cells, which I do not believe is opendp's fault.

I do wonder however if the above case should be flagged by opendp in one of it's checks.

jjcherry56, 2020-02-26:
To my knowledge there is no requirement for fixed placed cells to be
aligned with the rows.

However, the fact that the tapcells are not on the grid is a problem, but
not an opendp issue.
Apparently tapcell is supposed to respect the rows when placing the tap
cells and is not.
I suggest you file a different issue against tapcell with a testcase
showing exactly how you added them.

kareefardi, 2020-02-26:
Alright, thanks a lot for your help.

eder-matheus, 2020-02-27:
Hi, @kareefardi. You can find an example script for tapcell here: https://github.com/The-OpenROAD-Project/OpenROAD/blob/openroad/src/tapcell/etc/scripts/example_45nm.tcl.

This should be enough to have a properly tapcell insertion.