The-OpenROAD-Project/OpenROAD #134: pdngen: Non-standard order of layers in VIAS section

ax3ghazy, 2020-03-05:
Hello,

So the LEF/DEF reference defines the VIAS section in a DEF file as follows:

```
[VIAS numVias ; 
[– viaName  
   [   + VIARULE viaRuleName  
          + CUTSIZE xSize ySize  
          + LAYERS botmetalLayer cutLayer topMetalLayer  
          + CUTSPACING xCutSpacing yCutSpacing  
          + ENCLOSURE xBotEnc yBotEnc xTopEnc yTopEnc  
          [+ ROWCOL numCutRows NumCutCols]  
          [+ ORIGIN xOffset yOffset]  
          [+ OFFSET xBotOffset yBotOffset xTopOffset yTopOffset]  
          [+ PATTERN cutPattern] ] 
   | [ + RECT layerName pt pt | + POLYGON layerName pt pt pt] ...] 
;] ... 
END VIAS]
```
In the golden [file](https://github.com/The-OpenROAD-Project/OpenROAD/blob/484b8f0834d4fac5c5d49b6fb956b858f4589e8f/src/pdngen/test/gcd/gcd_pdn.check#L85) you're using, the order is bot-top-cut.

I have checked the source code and it seems that pdngen tries to output them in the right order (?), but I haven't dug deeper to see if this traces back to a problem with parsing.

Thanks.

Colin-Holehouse, 2020-03-05:
Hi,

the file you are referencing is not used as the golden reference – the file test_gcd.defok is the golden reference at the next level up.
The order of the layers in the VIAs for this file in the openroad branch is correct – the master will get updated from here in due course.
The existence of the gcd_pdn.check file is confusing, I will remove it.

Thanks.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

ax3ghazy, 2020-03-05:
Ah, alright; I just wanted to point out the issue on master using that file but wasn't aware it was fixed on the `openroad` branch.

Thanks!

closed 2020-03-05 by ax3ghazy