The-OpenROAD-Project/OpenROAD #136: support for tclreadline (or equivalent) ?

rbarzic, 2020-03-06:
Hi,

Not really an issue but it seems that I can't access command history through the up/down arrows.  My regular tclsh/wish shells have access to it through a specific ~/.tclshrc file but apparently the build-in tcl shell does not use it. Maybe there is an alternate solution available  or my build of OpenRoad is not supporting it because of some missign configuration ?

maliberty, 2020-03-06:
Its on my todo list but never quite reaches the top.  Its not automatic in tcl and I haven't quite found the magic to enable it.

jjcherry56, 2020-03-07:
as an emacs user I have no need for it (all shell windows have much better editing than tclreadline provides). there is an outstanding jira task to address it

arlpetergadfort, 2020-03-09:
Matt,

I have a basic tclreadline enabled on the most recent commit to the openroad branch (7ff2b23) that we are using at ARL.
I've attached the patch as a zip file [0001-Adding-tclreadline-init-to-OpenROAD.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/4307581/0001-Adding-tclreadline-init-to-OpenROAD.zip)

I tried to keep the changes to a minimum, so it should be relatively easy to verify/see the changes.
Please reach out to me if you have any issues.

This patch basically enables tab completion of proc names (but not options) and filenames, basic stuff.

maliberty, 2020-03-09:
I've received your patch.  I find that for my local system TCL_LIB_PATHS is empty and so tclreadline isn't found.  I'll have to spend a bit of time to see why it isn't working for me.  What platform did it work for you on?

arlpetergadfort, 2020-03-09:
I ran it on RHEL6, but had to first install the "tcl-tclreadline*" package with yum. The tcl-tclreadline library is located at /usr/lib64/tcl8.5/ (on RHEL6 on the TSS). Thanks for taking a look at it.
It's possible I did not configure the CMake file correctly in FindTCL.cmake and I'm not as familiar with CMake (especially the find_library command).

arlpetergadfort, 2020-03-09:
Looking through the FindTCL.cmake file, It might be more appropriate to use the TCL_PATH_DIR instead of the TCL_LIB_PATHS (which does appear to have a chance of being empty).

maliberty, 2020-03-30:
I get this error from cmake with your patch:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
TCL_READLINE_LIBRARY

I don't see where you have defined TCL_READLINE_LIBRARY.  Is something missing from the patch?

maliberty, 2020-03-31:
This has been integrated into openroad with some minor changes (https://github.com/The-OpenROAD-Project/OpenROAD/commit/1635259381e9534fa0bc0368b8616ce391fb53cf).

Thanks for the contribution.  It is on the 'openroad' branch which is the development head.

closed 2020-03-31 by maliberty