The-OpenROAD-Project/OpenROAD #138: opendp segmentation fault for padding

sewkim, 2020-03-08:
In opendp, if I assign -left or -right in set_placement_padding large enough to cause placement to fail, a seg fault occurs.
The following is the test code that causes seg fault by modifying only the corresponding function in the pad01.tcl test file.

`# set_padding -global -left
source helpers.tcl
read_lef Nangate45.lef
read_def simple01.def
# set_placement_padding -global -left 5
set_placement_padding -global -left 30
detailed_placement
check_placement

set def_file [make_result_file pad01.def]
write_def $def_file
diff_file $def_file pad01.defok`

Thanks

jjcherry56, 2020-03-10:
fixed in 94b3848 opendp issue 138

jjcherry56, 2020-03-10:
In the future, please to not copy files from the test directory and rename them. I reuse files in the test directory and do not want to have multiple copies of the same data. I shouldn't have to guess and diff to find your def is the same as one that was already there.

closed 2020-03-10 by jjcherry56

sewkim, 2020-03-10:
I see, I'll be more careful in future. Thank you for your response.