The-OpenROAD-Project/OpenROAD #140: OpenDP bug in filler_placement with set_placement_padding

sewkim, 2020-03-08:
[image]
[image]

The filler_placement command with set_placement_padding generated overlapped placement result.

There are overlapped cells after OpenDP.
I attached Innovus checkPlace results that contains a list of overlapped cells (bug02_result.rpt).

Could you check the attached testcase?

[bug02.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/4303858/bug02.tar.gz)

I used the following commit in OpenROAD repo:
https://github.com/The-OpenROAD-Project/OpenROAD/commit/4187ce74b7f7dccb126f9870383511eca24132b5

Thanks,

jjcherry56, 2020-03-10:
what happens if you remove the second contradictory call to set_placement_padding?
it make absolutely no sense to change the padding mid-flow

jjcherry56, 2020-03-10:
I see. It isn't filling the padding sites so you were trying a work around.
I think the real issue is not filling the padding.

jjcherry56, 2020-03-11:
Try it now (after commit 3150c4e). Thanks for the test case.

sewkim, 2020-03-11:
I confirmed that it worked fine. Thank you very much.