The-OpenROAD-Project/OpenROAD #151: tapcell overlap with endcap

kareefardi, 2020-03-12:
Hello,

Using tapcell tool to insert tapcells and decaps cells, depending on the input distance and the design size, can lead to overlapping tapcells with the right decap column. Please check with @tspyrou for a private repo for a test case. 

Thanks

tspyrou, 2020-03-16:
Eder, thanks for looking into this. I am talking to Kareem now and he was happy to see you assigned it to yourself.

eder-matheus, 2020-03-17:
Hi, @kareefardi.

I've updated tapcell at this [branch](https://github.com/The-OpenROAD-Project/OpenROAD/tree/fix/tapcell_overlap). It should not create overlaps anymore when using the script for 45nm.
The change for 14nm will come up soon.

Thanks.

tspyrou, 2020-03-26:
This branch has been integrated and this issue should be resolved now.

closed 2020-03-26 by tspyrou