The-OpenROAD-Project/OpenROAD #164: plist problems: Installs bundled depondencies and causes conflicts with preinstalled packages

yurivict, 2020-03-24:
The installed plist looks like this:
```
bin/openroad
bin/sta
include/ArcDelayCalc.hh
include/Arnoldi.hh
<...more OpenSTA headers...>
```
The above conflict with a separate ```opensta``` package.
```
include/lemon/adaptors.h
include/lemon/arg_parser.h
<...more lemon headers...>
```
these conflict with the separate ```lemon``` package.

```
/usr/ports/cad/openroad/work/.OpenPhySyn/transforms/libbuffer_fanout.a
/usr/ports/cad/openroad/work/.OpenPhySyn/transforms/libconstant_propagation.a
/usr/ports/cad/openroad/work/.OpenPhySyn/transforms/libgate_clone.a
/usr/ports/cad/openroad/work/.OpenPhySyn/transforms/libpin_swap.a
@dir /usr/ports/cad/openroad/work/.OpenPhySyn/transforms
@dir /usr/ports/cad/openroad/work/.OpenPhySyn
@dir /usr/ports/cad/openroad/work
@dir /usr/ports/cad/openroad
@dir /usr/ports/cad
@dir /usr/ports
```
The above files are installed into obviously wrong paths.

--------

The above problems result from bundling used in the project. Please consider having cmake options to use separate packages when they are available.

tspyrou, 2020-04-04:
This is good feedback. So far we have not had users installing the software on the system but in home directories since it is changing so much. We do want to make system installation work cleanly.

maliberty, 2020-04-08:
We are tending to move to not include third party packages in the code base but to make them external dependencies.  We just did that change with Eigen.  Would removing lemon from the code base resolve this concern?

yurivict, 2020-04-08:
Both eigen and lemon, and all other third party packages should be removed, and external packages used for them. Not sure if this would fix the overall issue. Need to try and see.

maliberty, 2020-04-08:
I changed the OpenPhySyn transforms to go into the install hierarchy.

maliberty, 2021-01-05:
OpenPhySyn has been removed from openroad so this no longer seems relevant.  Most 3rd party packages are now treated as separate installs.

closed 2021-01-05 by maliberty