The-OpenROAD-Project/OpenROAD #181: openroad build completed on Ubuntu; however, I get openroadd commanddd not found.

alaasal, 2020-04-01:
I tried to bulid openrod on ubuntu 18 the build completes successfully but when I run the command open road I get the error command openroad not found. 
[image]

maliberty, 2020-04-01:
I expect you don't have it in your path.  See if /home/user/OpenROAD-flow/tools/OpenROAD/build/src/openroad works.

rovinski, 2020-04-01:
Hi @alaasal,

It looks like you are using OpenROAD-flow. There is a script `setup_env.sh` which will set up your environment.

alaasal, 2020-04-01:
Thanks all, I didn't have the command in my path; however, sourcing setup_env.sh didn't solve that and I had to add openroad and tritonroute to my path myself.

rovinski, 2020-04-01:
Sourcing `setup_env.sh` should work unless you didn't use the `build_openroad.sh` script to build. If so, please file an issue against [OpenROAD-flow](https://github.com/The-OpenROAD-Project/OpenROAD-flow).

closed 2020-04-01 by rovinski