The-OpenROAD-Project/OpenROAD #196: Building openroad branch failed with a fatal error: tcl.h: No such file or directory

rbarzic, 2020-04-07:
openroad commit : 7fa1073654f5fbfdbaffa6cbab6102c6919d614c (branch openroad)

Steps to reproduce :
```
git clone --branch openroad --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git OpenROAD-debug
cd OpenROAD-debug
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=$HOME/tmp/openroad_install ..
make
```

cmake correctly locates the tcl lib and header:
```
 cmake -DCMAKE_INSTALL_PREFIX=$HOME/tmp/openroad_install .. | grep -i tcl
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
```

but build fails with error:

```
[100%] Building CXX object src/OpenSTA/CMakeFiles/sta_swig.dir/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx.o
cd /home/ronan/perso/github/OpenROAD-debug/build/src/OpenSTA && /usr/bin/c++  -DOPENROAD_BUILD -I/home/ronan/perso/github/OpenROAD-debug/src/OpenSTA/include/sta -I/home/ronan/perso/github/OpenROAD-debug/src/OpenSTA -I/home/ronan/perso/github/OpenROAD-debug/src/OpenSTA/include  -O3 -DNDEBUG   -std=c++11 -o CMakeFiles/sta_swig.dir/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx.o -c /home/ronan/perso/github/OpenROAD-debug/build/src/OpenSTA/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx
/home/ronan/perso/github/OpenROAD-debug/build/src/OpenSTA/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx:166:10: fatal error: tcl.h: No such file or directory
 #include <tcl.h>
```
Running cmake with TCL_HEADER option gives the same error

( cmake -DCMAKE_INSTALL_PREFIX=$HOME/tmp/openroad_install -DTCL_HEADER=/usr/include/tcl/tcl.h .. )

Apparently, the top level information about TCL file location is not propagated correctly down to OpenSTA

maliberty, 2020-04-07:
Ubuntu puts tcl.h in /usr/include/tcl which has caused us grief before.  I've proposed a fix internally to OpenSTA.

rbarzic, 2020-04-07:
Ok - But apparently something change during the last  7/10 days because I was able to build the openroad branch before

maliberty, 2020-04-07:
OpenSTA had a number of changes to its cmake recently.  Most development happens on Centos which doesn't have this issue, so it tends to go unnoticed until someone tries ubuntu.  That's often me but in the case you were first :)

rovinski, 2020-04-08:
@maliberty I would propose simply making an Ubuntu Dockerfile and adding an Ubuntu build to the Jenkins regressions. Then this type of thing shouldn't happen again.

rbarzic, 2020-04-08:
Apparently, switching to  the master branch of OpenSTA  seems to work while trying to build the openroad branch of OpenROAD

maliberty, 2020-04-08:
I agree about ubuntu but up until yesterday we didn't have the jenkins capacity to afford another platform.  Now that are using gcloud it should be possible.

arlpetergadfort, 2020-05-19:
I was able to get 0d16e26 to compile (the most recent used in OpenROAD-flow). For Ubuntu 20.04 LTS.

ABC also has this issue and a few simple changes to the Make files would solve this. I've attached the two patches if you (@maliberty) are interested in applying them to your forks of YOSYS and ABC. The YOSYS one needs to be edited to insert the correct commit hash for the new ABC commit.
1. [ABC-0001-Introduce-TCL_INCLUDE-flag-to-ensure-proper-build-on.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/4651363/ABC-0001-Introduce-TCL_INCLUDE-flag-to-ensure-proper-build-on.zip)
2. [YOSYS-0001-Ensuring-TCL_INCLUDE-flag-is-passed-to-ABC-build-so-.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/4651362/YOSYS-0001-Ensuring-TCL_INCLUDE-flag-is-passed-to-ABC-build-so-.zip)

For Ubuntu I needed to run the following command:
	`sudo apt-get install cmake gcc swig tcl-tclreadline libboost-dev libeigen3-dev`
Unfortunately, LEMON is currently not in the Ubuntu repos (for 20.04), so I had to install that by compiling it from the source.

maliberty, 2020-05-21:
Lemon is there for 18 & 19 so perhaps it will show up for 20 later on.  The joys of the bleeding edge.

I'll look at your other changes.

maliberty, 2020-05-21:
I've integrated your changes.  I was able to build yosys on Ubuntu 19.  I've been wanting to get to this for a while so thanks for the patches!

closed 2020-05-21 by maliberty