The-OpenROAD-Project/OpenROAD #197: Error message is not immediately actionable "cannot insert center cellbuffer1 target (17964050 8801100)"

oharboe, 2020-04-07:
This error message needs to be a bit more understandable.

Is it a bug in the detailed routing? Any hints as to what to fix?

I know the design fits...

```
Design area 393799 u^2 0% utilization.
openroad -no_init ./scripts/detail_place.tcl \
         2>&1 | tee ./logs/nangate45/AptosAvalonMMSlave/3_3_opendp.log
OpenROAD 1.1.0 
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Notice 0: Reading LEF file:  ./objects/nangate45/AptosAvalonMMSlave/merged_padded.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  ./objects/nangate45/AptosAvalonMMSlave/merged_padded.lef
Notice 0: 
Reading DEF file: ./results/nangate45/AptosAvalonMMSlave/3_2_place_resized.def
Notice 0: Design: AptosAvalonMMSlave
Notice 0: 		Created 100000 Insts
Notice 0: 		Created 200000 Insts
Notice 0: lines processed: 1000000
Notice 0: lines processed: 2000000
Notice 0: lines processed: 3000000
Notice 0: lines processed: 4000000
Notice 0: lines processed: 5000000
Notice 0: lines processed: 6000000
Notice 0: lines processed: 7000000
Notice 0:     Created 5070 pins.
Notice 0:     Created 253063 components and 569150 component-terminals.
Notice 0:     Created 2 special nets and 506126 connections.
Notice 0:     Created 27440 nets and 63024 connections.
Notice 0: Finished DEF file: ./results/nangate45/AptosAvalonMMSlave/3_2_place_resized.def
-------------------- Design Stats ------------------------------
core area                  : (40280, 44800) (17246680, 17236800)
total cells                : 253063
multi cells                : 0
fixed cells                : 230321
nets                       : 27442
design area                : 2.96e+14
total fixed area           : 2.45e+11
total movable area         : 3.42e+11
design utilization         : 0.116
rows                       : 6140
row height                 : 2800
----------------------------------------------------------------
Error: cannot insert center cellbuffer1 target (17964050 8801100)
Error: cannot insert center cellbuffer2 target (17964050 4924220)
...
[deleted]
Check Legality
row check ==> PASS 
site check ==> PASS 
power check ==> PASS 
edge_check ==> PASS 
placed_check ==> FAIL (1461)
overlap_check ==> FAIL 
-------------------- Placement Analysis ------------------------
total displacement         : 1711741403
average displacement       : 6764
max displacement           : 35950830
original HPWL              : 2.21e+07
legalized HPWL             : 5.08e+07
delta HPWL                 : 130%
----------------------------------------------------------------
Error: cannot insert center cellbuffer1 target (17964050 8801100)
...
[deleted]
make: *** [results/nangate45/AptosAvalonMMSlave/3_3_place_dp.def] Error 1

real	96m51,083s
user	0m0,873s
sys	0m0,627s
```

maliberty, 2020-04-07:
Looks like opendp but I'm not certain.  Hopefully Cherry will have some idea.

rovinski, 2020-04-08:
Based on the log format, this appears to be using a version of openDP that isn't on the head of the openroad branch, probably master. Cherry has fixed a number of openDP issues in the openroad branch.

oharboe, 2020-04-08:
@tajayi @rovinski  I'm using the tip of OpenROAD-flow master branch. How is OpenROAD-flow updated? Is there an automated procedure that is run regularly?

Are there some regression tests run pre-merge on updates to OpenROAD-flow?

It would seem to me that it can be quite tricky to make sure that none of the tips of the other sub-projects don't introduce a subtle regression...

oharboe, 2020-04-08:
How can I re-build OpenROAD-flow correctly with the openroad branch?

rovinski, 2020-04-08:
@oharboe OpenROAD-flow/openroad branch is tied to OpenROAD/openroad and OpenROAD-flow/master is tied to OpenROAD/master (or at least they are in theory). OpenROAD-flow/master has been stuck for several weeks because it is failing some tests on our large-scale regressions.

We do small-scale regression tests on the openroad branch and then large-scale regressions (which take about 14-18 hours to run) before merging into master. We have this large/small breakdown because it is not feasible to run 14 hours worth of regressions on every commit.

If you want to build OpenROAD-flow with the openroad branch, you can simply git checkout OpenROAD-flow openroad branch. In my opinion, "openroad" branch is a bit of a misnomer because it is equivalent to the dev branch or nightly branch of other open source projects.

oharboe, 2020-04-08:
@rovinski  Scaling pre-merge tests is a fascinating topic and extremely well studied and solutions exist.

When I worked on geoscience software, we had 60000 hours worth of tests/day if we had tried to brute force checked all incoming pull requests with all tests for the whole team(team of 50 developers all producing a couple of pull requests and a big datasets in tests). The way we broke the back of the problem there was to create statistics of what tests where the most valuable and quickest. We then ran those tests on each pull request. The turnaround for this testing was ca. 1 hr. Then we merged in all incoming pull requests for the day and ran them overnight on the full test set. If that passed, the pull requests were automatically merged. So that's the equivalent of 60000 hours of tests per day.

I've implemented similar systems and test scaling pre-merge since.

For OpenROAD, something similar should be eminently doable, and in fact quite necessary, if you are to match the infrastructure to your ambitions of creating a fully automated flawless custom silicon toolchain with teams cooperating across the world.

rovinski, 2020-04-10:
@oharboe we pretty much already do this. The problem is that little attention is paid to the master branch so when the large-scale tests break, developers continue to focus on the openroad (development) branch. I'm an advocate for stability first and productivity second, but I'm in the minority. Most developers are focused on meeting the goals set by our project sponsor first, because failure to do so would likely result in losing our funding and the project dying.

oharboe, 2020-04-10:
@rovinski Exactly. You're missing the last bit, the magic sauce, to make this work. The integration test has to be implemented in a fashion that scales and it has to be run on all pull requests prior to merge. Nobody should be merging with the master branch manually, it has to be automatic. If something breaks, there should simply be a pull request that hasn't merged yet. The master branch should always work and images for betatesters should be ready overnight automatically.

oharboe, 2020-04-10:
@rovinski A technical point: I predict you can't get to where you need to be with git submodule. I think what you need is git subtree.

With git subtree you have:

1. A single giant repository. 
2. The ability to merge in from upstream and contribute back to upstream to e.g. Yosys.
3. Since you have a single giant repository, the github pull request build infrastructure works out of the box. Create a branch that modifies multiple projects, test it all as a single pull request and makes ure it all works together before merging automatically.
4. Even if you have a single giant git repository, the projects are distinct.
5. The cognitive load is much lower for developers who just want to get stuff done and doesn't want to worry about how their sub-project fits into the scheme of things.

maliberty, 2020-04-10:
The single repo is a big issue with competing viewpoints; subtree is one possible answer.   The fully automatic merge is a good idea but right now there is nobody to work on such CM infrastructure.  We will get there eventually but I think this is drifting from a bug report into a philosophy discussion.

oharboe, 2020-04-10:
@maliberty Indeed. For now, I guess with waiting to retest until OpenROAD-flow master is updated.

oharboe, 2020-04-11:
This is fixed on the openroad branch once https://github.com/The-OpenROAD-Project/OpenROAD-flow/pull/49 is merged as well.

closed 2020-08-18 by maliberty