The-OpenROAD-Project/OpenROAD #198: Synthesis of small circuits

RatkoFri, 2020-04-07:
Greetings, 

I am trying to synthesize small circuits like multipliers. In one tool (QFLOW)  my designs exhibit really small energy consumption. On the other hand OpenROAD, I get totally different results. Can you give some tips on how to efficiently synthesize small designs?

Ratko

rovinski, 2020-04-08:
Hi @RatkoFri,

Can you please be more specific about the behavior you are seeing? Can you provide us with the verilog and design configuration inputs? Does the difference show up when targeting the nangate45 library, and if so, can you provide the specific numbers you are seeing?

rovinski, 2020-04-08:
Closing as duplicate of https://github.com/The-OpenROAD-Project/OpenROAD-flow/issues/45

closed 2020-04-08 by rovinski