The-OpenROAD-Project/OpenROAD #205: Release?

FFY00, 2020-04-13:
Hello, I am interested in packaging OpenROAD. This requires for a release to exist, so I was wondering if you have any plans about this? Perhaps a timeline?

Thanks,
Filipe

maliberty, 2020-04-15:
We are tentatively planning a 1.0 release around Q3.  What sort of packing do you have in mind?

FFY00, 2020-04-15:
I want to package this for Arch Linux.

sequencer, 2020-05-13:
@FFY00 a git version has been packages in aur.

FFY00, 2020-05-13:
The AUR is not a package repository, it is a PKGBUILD repository. This means users will have to build the package themselves and it may break anytime due to untracked soname changes. Having it in the official repositories should be much better.

sequencer, 2020-05-13:
Yes, we have to wait for a release for OpenROAD before packaging it into community. Maybe I can push it to the archlinuxcn repository firstly to get rid of the rebuilding burden. After we get the first release, it can upstream into official repository.

closed 2021-11-28 by maliberty