The-OpenROAD-Project/OpenROAD #206: filler_placement places cells on top of macros

rovinski, 2020-04-14:
When I run filler_placement, I see that fill cells get placed on top of block macros:
[image]

These sites should not be filled because they have the cell sites removed:
[image]

Test case: 
[fillcell_tinyRocket_nangate45_2020-04-14_17-13.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/4477928/fillcell_tinyRocket_nangate45_2020-04-14_17-13.tar.gz)

rovinski, 2020-05-02:
This was resolved at some point.

rovinski, 2020-05-11:
This issue has resurfaced (or perhaps never went away and I was looking at the wrong view). The same case is valid.

jjcherry56, 2020-05-14:
fixed by commit fe43ab7 opendp no fillers over blocks

closed 2020-05-14 by jjcherry56