The-OpenROAD-Project/OpenROAD #215: [pdngen] stripes not exposed as pins for LVS and power routing

ax3ghazy, 2020-04-20:
Hello,

This is a feature needed for two main necessary purposes: LVS (to verify that all macros are powered) and power routing (power pins need to be exposed when a LEF is generated from the final routed DEF to be used in bigger designs).

Currently, both of these steps require manual intervention. Creating 2 0-area pins on top of a VDD and VSS stripes for the purpose of LVS does the job. Power routing is done entirely by hand (let me know if you need further details on this).

To eliminate these, in my opinion, (at least two) stripes should be exposed as power pins or perhaps extended to the macro boundary. Power routing could either (perhaps) be implemented as a special feature from FastRoute+TritonRoute (by specifying the required widths and metal layers) or by through a new independent tool.

Let me know if you need further details,
Thanks

Colin-Holehouse, 2020-04-20:
I agree, need to make the top layer power straps be pins in a block level design.

The most straight-forward way to add this will be to have the highest metal straps in a block design be assigned as pin shapes as well as net shapes. This will allow power routing from the top level to drop via connections onto each strap as it routes over the block, making for a robust power grid. I prefer this to extending the metal straps to the edge of the block.

Please let me know if you see any significant issues with this approach.

ax3ghazy, 2020-04-20:
Thanks for the quick response.

I agree that this approach would solve the issue in many cases where a top-level design does indeed have a "compatible" PDN, onto which the PDNs of the contained blocks can be hooked. However, with a metal stack of only 4-5 metal layers, it would probably be the case that the stripes of the top-level macros and those of the blocks inside are on the same metal layer. Would that approach work in such a case?

Also, in cases where there is actually no PDN on the top-level but the blocks are rather enclosed with a padframe, power routing still needs to be handled externally: e.g., using a FR+TR solution or an independent solution to route power from the power pads to the PDN of the blocks; is that correct?

Would having a power ring in the PDN, exposed externally as pins solve the issue? That's what was being done manually by hand so far, and it seems to me like a simple solution in both of the above cases, namely to either hook up to a top-level PDN or be easily routed to power pads (now it would just need a pair of wide wire segments on a high metal layer from the power pads to the power ring). Would such a solution make sense?

Thanks!

Colin-Holehouse, 2020-04-21:
Ok, so for processes with a high enough number of metal layers we can be happy with a strategy that uses a limited number of layers (no more than TOP-1) in the sub blocks (BLOCK_TOP <= TOP-1), with metal straps on the BLOCK_TOP layer being promoted to have pin shapes that can be connected from above.

I'd like to explore the other cases a bit more fully to try to come to an understanding of what the right strategy should be.

I should point out first that there is the ability to add core power/ground rings as part of the power grid and automatically hook these up to named pad cell types. (See: https://github.com/The-OpenROAD-Project/OpenROAD/blob/openroad/src/pdngen/doc/PDN.md - in the section "Additional features for top level power grid"), I believe this should serve to bridge the power connections from the pad cells to the top power grid. In case it's not clear from the doc, connections from the core rings to padcells take precedence over connections from the corerings to the main power grid.

I'm not sure if this helps with the handling of sub-blocks, we could have core rings be created inside a sub-block and add some code to promote them as pins, but there would need to be a complimentary update to pdngen to be able to connect to "ringed" blocks, as this isnt something that is currently handled.

Seems most likey that for the small number of layers situatin we will have to resort to extendeding layers BLOCK_TOP and BLOCK_TOP-1 to the edges of the block, where it would then be expected to line up with top level stripes, or else connect to power rings around the outside of blocks.

Do any of these ideas line up with your current practices for these situations? I'm hoping that you can describe your current approach(es) for connecting to sub-blocks in processes with limited metal layers, and that will help me gain some insight into what is needed for the openroad code.

ax3ghazy, 2020-04-27:
Sorry for the delay.

> Ok, so for processes with a high enough number of metal layers we can be happy with a strategy that uses a limited number of layers (no more than TOP-1) in the sub blocks (BLOCK_TOP <= TOP-1), with [...]

Agreed.

> I should point out first that there is the ability to add core power/ground rings as part of the power grid and automatically hook these up to named pad cell types.

This is already a good solution for one of the main top-level integration approaches, namely having a single "super macro" at the top-level. I have checked the black parrot .defok file, but it's there to test the features you mentioned not to be a complete example by itself, right?

Another case of top-level integration is when there are multiple macros at the top-level, each having an own PDN. In this case, a possible solution (at a high-level) may go as follows. Assuming that all macros are placed such that their PDNs have **aligning** straps, these straps would be extended and a top-level grid with the same pattern created over the core area inside the padframe. Then a core ring would be created, just like in the black_parrot example, to which the straps will be connected as well as the power pins of the pads. 

> I'm not sure if this helps with the handling of sub-blocks, we could have core rings be created inside a sub-block and add some code to promote them as pins, but there would need to be a [...]

> or else connect to power rings around the outside of blocks.

These sound like a very reasonable solution for handling sub-blocks to me.

Colin-Holehouse, 2020-05-04:
If you have any comments about how the .defok looks in the current example for core rings, then I would find it insightful to hear them. I can see that I could adapt the approach that I'm using to connect padcells to the core ring to do the same thing from the single core macro to the core rings.

If you have macros with aligning straps then you could insert a regular power grid over the core area and define the macro strategy such that the blocks do not obstruct the power straps. My experience with this approach in the past is that it's hard to enforce such a rigorous approach especially if the blocks are being done by different team members.

I have some code to add p/g pins for a block, however, currently the underlying database API only supports a single shape for each pin. I've requested an API update and when the API is done, I can activate the code to generate the pins in the DEF output.

I will also have a think about extending block pins to meet the core rings. I'm open to any further suggestions for P/G strategies that you think may be usefully implemented.

maliberty, 2020-08-31:
@Colin-Holehouse is there something required from OpenDB for this issue?

maliberty, 2020-11-28:
@Colin-Holehouse any update on this?

maliberty, 2020-12-16:
@Colin-Holehouse opendb now supports multiple shapes (dbBox) per dbBPin.  This should enable you to the put the grid in a single pin.