The-OpenROAD-Project/OpenROAD #253: Create a RPC interface as an optional replacement to tcl

sequencer, 2020-05-13:
Currently, if user wanna interact with OpenROAD, the only way is to invoke a tcl shell to send command and receive command. The result is hard to parse, and string only without type struct informations.
So I wonder if possible to add a json/protobuf based RPC interface as a optional replacement to tcl which contains a more serializable API for other program to interact with OpenROAD.

maliberty, 2020-05-18:
What would this interface look like?  All the command parsing is done in tcl today.

sequencer, 2020-05-23:
I think it can be something like a daemon, which has can consume a json input and generate a json out put, and has a standard serialise and deserialise specification which can be implemented in other program too.

closed 2020-11-09 by sequencer