The-OpenROAD-Project/OpenROAD #269: Very long runtime for the report_checks command

killruana, 2020-05-19:
**Describe the bug**
I try to implement an opensource FPGA with the openroad flow.
The report_checks command of the floorplan step was still not finished after 48 hours of execution.

The design is not so big (120k standard cells) but it contains many combinational feedback loops. Maybe this is what the STA tool doesn't like?

**Expected behavior**
A short runtime :)

**Environment (please complete the following information):**
 - OS: Archlinux (I know, not officially supported)
 - OpenROAD commit c60f97245dccd6963a459ede934c7189b3bba86e
 - OpenROAD-flow commit 229a1f41fc9bae5f5dfd871c799edceb985d4b7e

**File Uploads**
The flow is available here https://git.slaanesh.org/killruana/openroad-flow-k1g100/src/branch/k1g100

```
$ git clone https://git.slaanesh.org/killruana/openroad-flow-k1g100
$ cd openroad-flow-k1g100
$ git checkout k1g100
$ cd flow
$ make DESIGN_CONFIG=designs/nangate45/k1g100.mk
```
I also join an archive: 
[k1g100.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow/files/4568515/k1g100.tar.gz)

**Additional context**
Add any other context about the problem here.

jjcherry56, 2020-05-20:
If you use the make target floorplan_issue the flow will package the files necessary to run the floorplan step without the flow (see util/util.mk). This is how issues are handed off to openroad. k1g100.tar doesn't really help because it is rtl. I am waiting for yosys to do its thing.

jjcherry56, 2020-05-20:
Austin packaged the issue (I didn't notice it because it was filed in a different bug system).
Yes, the issue is combinational paths. It is stuck trying to levelize the design.
Is there a "mission mode" you can enable with set_disable_timing on some feedback paths?
OpenSTA disabled 1472 edges but still finds combinational paths 7300 gates long in the design.

tspyrou, 2020-05-21:
Hi Jonathan,

I am happy to see that you are using OpenROAD and also that you are working on an open source FPGA architecture. As a background, I was the technical leader for Device modeling and Timing Analysis in Quartus at Altera/Intel for 8 years before joining OpenROAD last year. I am very familiar with the problem that you are running into.

I would recommend that you run the OpenROAD flow without timing constraints as a start. You will get through the flow without the path explosion that Cherry mentions. Timing analysis of an un-programmed FPGA is not yet possible even in commercial STA tools. There are too many possible paths in the un-programmed logic. I worked on this problem for a few years in the past and am very familiar with STA tool limitations here and possible workarounds and solutions.

One solution which works well is that once you have completed non timing driven implementation, you can then use set_case_analysis to set the bits in your FPGA like a compiler would do and then do timing analysis of several use cases to see the performance. For hold analysis, for each register, you can use netlist traversal code at the TCL level to disable paths more then a few levels deep. Then you can report_checks -hold and ensure the design is hold clean without any program, ensuring the design is always hold clean regardless of program. FPGA's are slow and normally a few levels us much larger than any clock skew/OCV.

Another option is to make a very small section of the FPGA and see if timing analysis on it can complete. Even if it does complete most of the paths displayed will be false paths. Once you start setting set_false_path exceptions to block the false paths, you will find that you have to set so many that you will get tired of it or the tool will begin to slow down again.

The number of paths in a circuit is exponential and the logic depth is the exponent. STA tools are very good at turning this exponential problem into one that takes linear time. There are limits and un-programmed FPGA timing analysis is still beyond what is possible today in any STA tool.

Thanks and keep us posted with your progress!

Tom

tspyrou, 2020-05-21:
I should mention that we will likely close this issue. This does not mean that we are uninterested to help as possible.

Tom

jjcherry56, 2020-05-21:
To disable the STA you will need to modify the flow scripts to remove report_checks, report_check_types and resizing commands. macro_placement also depends on timing
so I hope you don't have macros in the design.

killruana, 2020-05-23:
Thank you for your responses!

Now that I have confirmation it is an issue with my design and not the tool, feel free to close the issue.

closed 2020-05-27 by rovinski

jjcherry56, 2021-05-19:
I recently found and fixed an issue that may have prevented your design from running. If it still matters you could try again.