The-OpenROAD-Project/OpenROAD #299: [TritonCTS] Exiting on trivial designs with 1 sink

rovinski, 2020-06-03:
TritonCTS currently skips nets which have 1 sink, which makes sense because a tree cannot be constructed with 1 sink:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/771b124831043ab352bcfcb6b9da26ea24bdc81d/src/TritonCTS/src/DbWrapper.cpp#L159-L162

However, for trivial designs which have only 1 sink, this results in no clock nets being found and TritonRoute will exit on error:

https://github.com/The-OpenROAD-Project/OpenROAD/blob/771b124831043ab352bcfcb6b9da26ea24bdc81d/src/TritonCTS/src/DbWrapper.cpp#L110-L114

TritonCTS should be able to handle this case. I think the main thing is that TritonCTS should never call `std::exit`, it should return control to the main openroad app and allow the app decide what to do.

Even if TritonCTS returns and does nothing (other than issue WARNs and/or ERRORs), this is a tautological case that should not cause OpenROAD to exit.

Here is an example design (credit to @ax3ghazy):
```
module andd
(
  input  wire clk,
  output reg c,
  input  wire a,
  input  wire b,
  input wire reset
);

    wire x;
    assign x = a & b;

always @(posedge clk)
    if(reset)
        c <= 1'b0;
    else
        c <= x;
endmodule
```

And the resulting log:
```
 ************************
 *  Populate TritonCTS  *
 ************************
 Initializing clock nets
 Number of user-input clocks: 1 ( "clk" )
 Looking for clock nets in the design
 Net "clk" found
clk
    [WARNING] Net "clk" has 1 sinks. Skipping...
    [ERROR] No valid clock nets in the design. Exiting...
```

rovinski, 2020-06-03:
Let me know if you need a packaged test case; I can provide one.

rovinski, 2020-06-03:
Great, I think should work. If it's possible to return to the TCL interface with a non-zero status, that would be great so that it be handled in TCL as well.

jjcherry56, 2020-06-05:
See OR-314. I guess the part about not calling exit hasn't sunk in.

MarceloDanigno, 2020-06-05:
James, I updated all CTS code to not use exit anymore. It is in the new PR. (linked to this issue.) I didn't add it in the last PR (294) because I wasn't sure if it included C++ exits, since it is a little harder to treat some errors there (C++) than TCL. Sorry about that.

Austin, I based myself on opendp and pdngen to handle the errors. It works fine, but let me know if there is a better way for TritonCTS.

Again, thank you both for the reports.

MarceloDanigno, 2020-06-11:
I reviewed the PR and merged it into openroad just now. @rovinski in the testcase you mentioned, does the problem persist?

MarceloDanigno, 2020-06-15:
So, because of time constraints of the tapein, I'll go ahead and close this for now since it works with no problems in one of TritonCTS unit tests. You can find it below (modified to print a "Hello!" just after TritonCTS):
Script:
```
read_lef` "merged.lef"
read_liberty "merged.lib"
read_def "aes.def"
read_verilog "aes.v"
read_sdc "aes.sdc"

clock_tree_synthesis -lut_file "lut.txt" \
                     -sol_list "sol_list.txt" \
                     -root_buf CLKBUF_X3 \
                     -wire_unit 20 \
                     -clk_nets "clk" 

puts "Hello!"

exit
```
Report:
```
Notice 0: Reading LEF file:  merged.lef
Notice 0:     Created 22 technology layers
Notice 0:     Created 27 technology vias
Notice 0:     Created 134 library cells
Notice 0: Finished LEF file:  merged.lef
Notice 0:
Reading DEF file: aes.def
Notice 0: Design: aes_cipher_top
Notice 0:     Created 388 pins.
Notice 0:     Created 19005 components and 100029 component-terminals.
Notice 0:     Created 18831 nets and 61489 connections.
Notice 0: Finished DEF file: aes.def
 *****************
 * TritonCTS 2.0 *
 *****************
 *****************************
 *  Import characterization  *
 *****************************
 Reading LUT file "lut.txt"
    Min. len    Max. len    Min. cap    Max. cap   Min. slew   Max. slew
           2           8           1          52           1          24
    [WARNING] 180 wires are pure wire and no slew degration.
    TritonCTS forced slew degradation on these wires.
    Num wire segments: 4994
    Num keys in characterization LUT: 1677
    Actual min input cap: 8
 Reading solution list file "sol_list.txt"
 **********************
 *  Find clock roots  *
 **********************
 Running TritonCTS with user-specified clock roots: clk
 ************************
 *  Populate TritonCTS  *
 ************************
 Initializing clock nets
 Looking for clock nets in the design
 Net "clk" found
 Initializing clock net for : "clk"
    [WARNING] Net "clk" has 0 sinks. Skipping...
Error: No clock nets have been found.

Hello!
```
If the problem persists with the design from @ax3ghazy, contact me and I'll fix it ASAP (shouldn't be a difficult modification).

closed 2020-06-15 by MarceloDanigno

jjcherry56, 2020-06-15:
Per the round table discussion today, please write a test that has 1
instance in it instead of using aes with 13k instances.

ax3ghazy, 2020-06-15:
@MarceloDanigno 
> If the problem persists with the design from @ax3ghazy, contact me and I'll fix it ASAP (shouldn't be a difficult modification).

Sorry, do you have me confused with Ahmed Agiza (@ahmed-agiza)?

rovinski, 2020-06-16:
@ax3ghazy We had a user `ahegazy` report this problem on gitter, I thought it was you, but maybe it's just a coincidence.

@MarceloDanigno I tested the design and it looks fine. As @jjcherry56 mentioned, you should make sure this becomes a test case. It should be simple to do.