The-OpenROAD-Project/OpenROAD #358: clock_tree_synthesis -buf_list is not a known keyword or flag

gkamendje, 2020-06-30:
While trying to generate the characterization files,
I use the following command in a tcl script
clock_tree_synthesis -buf_list "$::env(CTS_BUF_CELL)" \
-sqr_cap 3.5e-20 \
-sqr_res 2.0e-4 \
-root_buf "$::env(CTS_BUF_CELL_ROOT)" \
-max_slew 50.0e-12 \
-max_cap 150.0e-15 \
-slew_inter 1.0e-12 \
-cap_inter 1.0e-15 \
-wire_unit 20 \
-clk_nets "clk" \
-out_path "$::env(CTS_TECH_DIR)" \
-only_characterization 0

as described in https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/TritonCTS.
However, I do get the following error

Error: clock_tree_synthesis -buf_list is not a known keyword or flag.

Here is the version of Openroad I am using and the output of help clock_tree_synthesis

OpenROAD 1.1.0 9295a533ff
% help clock_tree_synthesis
clock_tree_synthesis
[-lut_file lut] [-sol_list slist] [-root_buf buf] [-wire_unit unit] [-clk_nets nets]

So it looks like the option to generate the characterization files was not enabled in my version of
Openroad. How could I enable it?
Thanks

maliberty, 2020-06-30:
That commit is from Mar 31.  Please try the head version of master or openroad (the development branch).

gkamendje, 2020-07-01:
Hello,
Could you please help me with the specific version I should try? I checked out the openroad branch of both the flow as well as the openroad tool using the following:
```
 git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow
 cd OpenROAD-flow
 git checkout openroad
 git pull  origin openroad
 cd tools/
 cd OpenROAD/
 git checkout openroad
 git pull origin openroad
 cd ..
 ./build_openroad.sh
```
However, I get the following error 

```
-- Detecting C compile features - done
CMake Error at src/replace/CMakeLists.txt:25 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.
-- Configuring incomplete, errors occurred!
```
here are some git log output
from the flow dir: 

```
commit b31475af60a63ab92af5cc2e91cb8841c36fe163 (HEAD -> openroad, origin/openroad)
Merge: d2e2cab 30bcac4
Author: Matt Liberty <matt.liberty@gmail.com>
Date:   Tue Jun 30 10:26:14 2020 -0700
```
From the tools/OpenRoad directory

```
commit f7e4ab976ccaddef9414a2719ebdaec2f91485e6 (HEAD)
Merge: b830f989 484ab706
Author: James Cherry <34749589+jjcherry56@users.noreply.github.com>
Date:   Mon Jun 29 10:33:45 2020 -0700

    Merge pull request #355 from The-OpenROAD-Project/resizer

    Resizer

```

Any hint on what I am doing wrong would be appreciated. 
G

rovinski, 2020-07-02:
@gkamendje The error means exactly what it says. Eigen3 is a dependency of OpenROAD. Check the README for the required dependencies.

When using OpenROAD-flow, you do not need to do any cloning in the `tools` directory. `build_openroad.sh` takes care of that, and in fact will override whatever branch you try to clone in the tools folder. Use `build_openroad.sh --help` to view build options.

gkamendje, 2020-07-02:
@rovinski thanks for the hint. adding the the missing Eigen3 dependency and compiling with GCC 8.4.0 solved the issue. 
Thanks

closed 2020-07-02 by gkamendje