The-OpenROAD-Project/OpenROAD #360: OpenROAD/src/OpenPhySyn/src/FileUtils.cpp:34:10: fatal error: filesystem: No such file or directory

gkamendje, 2020-07-01:
Using the latest from GIthub I get the following error while trying to compile Openroad in a stand alone mode:
OpenROAD/src/OpenPhySyn/src/FileUtils.cpp:34:10: fatal error: filesystem: No such file or directory

```
commit 771b124831043ab352bcfcb6b9da26ea24bdc81d (HEAD -> master, origin/master, origin/HEAD)
Merge: 47859fed 54d8b1ca
Author: tspyrou <46945466+tspyrou@users.noreply.github.com>
Date:   Sat May 30 13:52:08 2020 -0700
```

How to reproduce:
```
git clone --recursice https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenROAD
 mkdir build
 cd build/
 cmake ..
 make

```

I have the followring prerequisites installed. 
  * cmake 3.14
  * gcc 8.3.0 or clang
  * bison 3.0.5
  * flex 2.6.4
  * swig 4.0

Libraries
  * boost 1.68
  * tcl 8.6
  * zlib
  * eigen
  * lemon
  * CImg (optional for replace)
  * spdlog (optional for OpenPhySyn)
Thanks for any hint

rovinski, 2020-07-02:
This problem should not happen if you are using GCC >= 8.3.0.

gkamendje, 2020-07-02:
@rovinski Thanks for the hint. I was able to succesfuly compile and link using GCC 8.4.0 with the options
` -DCMAKE_C_COMPILER=/usr/bin/gcc -DCMAKE_CXX_COMPILER=/usr/bin/g++`

However, linking fails when trying to compile with CLANG which is the default compiler
 
```
/usr/bin/ld: ../../src/OpenSTA/app/libOpenSTA.a(DispatchQueue.cc.o): undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
clang: error: linker command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/openroad.dir/build.make:209: recipe for target 'src/openroad' failed
make[2]: *** [src/openroad] Error 1
make[2]: Leaving directory '/home/gkamendje/tmp_compile/OpenROAD/build'
CMakeFiles/Makefile2:1043: recipe for target 'src/CMakeFiles/openroad.dir/all' failed
make[1]: *** [src/CMakeFiles/openroad.dir/all] Error 2
make[1]: Leaving directory '/home/gkamendje/tmp_compile/OpenROAD/build'
Makefile:152: recipe for target 'all' failed
make: *** [all] Error 2
```

After adding  -DCMAKE_CXX_FLAGS=-pthread to cmake  still with CLANG, I get the following error. 

```
OpenPhySyn/libOpenPhySyn.a(FileUtils.cpp.o): In function `psn::FileUtils::createDirectory(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
FileUtils.cpp:(.text+0x83): undefined reference to `std::filesystem::create_directory(std::filesystem::__cxx11::path const&)'
OpenPhySyn/libOpenPhySyn.a(FileUtils.cpp.o): In function `psn::FileUtils::readDirectory(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
FileUtils.cpp:(.text+0x1ee): undefined reference to `std::filesystem::__cxx11::directory_iterator::directory_iterator(std::filesystem::__cxx11::path const&, std::filesystem::directory_options, std::error_code*)'
FileUtils.cpp:(.text+0x334): undefined reference to `std::filesystem::__cxx11::directory_iterator::operator*() const'
FileUtils.cpp:(.text+0x42c): undefined reference to `std::filesystem::__cxx11::directory_iterator::operator++()'
OpenPhySyn/libOpenPhySyn.a(FileUtils.cpp.o): In function `std::filesystem::__cxx11::path::path<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::filesystem::__cxx11::path>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::filesystem::__cxx11::path::format)':
FileUtils.cpp:(.text._ZNSt10filesystem7__cxx114pathC2INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES1_EERKT_NS1_6formatE[_ZNSt10filesystem7__cxx114pathC2INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES1_EERKT_NS1_6formatE]+0x9b): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
OpenPhySyn/libOpenPhySyn.a(FileUtils.cpp.o): In function `std::filesystem::__cxx11::path::operator/=(std::filesystem::__cxx11::path const&)':
FileUtils.cpp:(.text._ZNSt10filesystem7__cxx114pathdVERKS1_[_ZNSt10filesystem7__cxx114pathdVERKS1_]+0x18): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
FileUtils.cpp:(.text._ZNSt10filesystem7__cxx114pathdVERKS1_[_ZNSt10filesystem7__cxx114pathdVERKS1_]+0x31): undefined reference to `std::filesystem::__cxx11::path::has_filename() const'
FileUtils.cpp:(.text._ZNSt10filesystem7__cxx114pathdVERKS1_[_ZNSt10filesystem7__cxx114pathdVERKS1_]+0x188): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/openroad.dir/build.make:209: recipe for target 'src/openroad' failed
make[2]: *** [src/openroad] Error 1
make[2]: Leaving directory '/home/gkamendje/tmp_compile/OpenROAD/build'
CMakeFiles/Makefile2:1043: recipe for target 'src/CMakeFiles/openroad.dir/all' failed
make[1]: *** [src/CMakeFiles/openroad.dir/all] Error 2
make[1]: Leaving directory '/home/gkamendje/tmp_compile/OpenROAD/build'
Makefile:152: recipe for target 'all' failed
make: *** [all] Error 2
```
I am closing this issue since I am able to compile and link with GCC but I think it would be great it someone could look at the CLANG linking issue. 
I am using Ubuntu 18.04 LTS
G

closed 2020-07-02 by gkamendje