The-OpenROAD-Project/OpenROAD #381: Need global mechanism to set TCL_LIBRARY & TCL_HEADER

20Mhz, 2020-07-15:
Hi, 
I've been having trouble building on macOS require a couple of work arounds. Each time I need to manually hack main build script to add -D TCL_LIBRARY=/opt/local/lib -D TCL_INCLUDE_PATH=/opt/local/include arguments to cmake 

Additionally, for OpenSTA some linking commands that are incomplete missing -ltcl:

/Library/Developer/CommandLineTools/usr/bin/c++  -O3 -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names  CMakeFiles/sta.dir/app/Main.cc.o  -o app/sta  app/libOpenSTA.a sta_swig.a /usr/lib/libz.dylib /usr/local/lib/libcudd.a app/libOpenSTA.a
fixed by appending to CMakeFiles/sta.dir/link.txt

20Mhz, 2020-07-15:
This other one is also missing libcudd.a:

cd /Users/ronaldv/Projects/repositories/msaligane/OpenROAD-flow/tools/build/OpenROAD/src && /usr/local/Cellar/cmake/3.17.3/bin/cmake -E cmake_link_script CMakeFiles/openroad.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++  -O3 -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names  CMakeFiles/openroad.dir/OpenRoad.cc.o CMakeFiles/openroad.dir/Error.cc.o CMakeFiles/openroad.dir/OpenRoadTclInitVar.cc.o CMakeFiles/openroad.dir/Main.cc.o  -o openroad  init_fp/src/libinit_fp.a openroad_swig.a replace/libreplace.a opendp/libopendp.a resizer/src/libresizer.a ../libioPlacer.a flute3/libflute.a dbSta/src/libdbSta.a /Users/ronaldv/Projects/repositories/msaligane/OpenROAD-flow/tools/OpenROAD/src/OpenSTA/app/libOpenSTA.a OpenDB/src/db/libopendb.a opendb_swig.a OpenRCX/src/libOpenRCX.a TritonCTS/libTritonCTS.a FastRoute/libFastRoute.a tapcell/src/libtapcell.a TritonMacroPlace/libTritonMacroPlace.a TritonMacroPlace/module/ParquetFP/libParquetFP.a TritonMacroPlace/module/ABKCommon/libABKCommon.a PDNSim/src/libpdnsim.a OpenDB/src/zlib/libzlib.a OpenDB/src/tm/libtm.a OpenDB/src/defin/libdefin.a OpenDB/src/defout/libdefout.a OpenDB/src/lefin/liblefin.a OpenDB/src/lefout/liblefout.a antennachecker/src/libantennachecker.a OpenDB/src/zutil/libzutil.a OpenPhySyn/libOpenPhySyn.a /usr/lib/libz.dylib ../libMunkres.a OpenRCX/src/openrcx-swig.a OpenDB/src/db/libopendb.a OpenDB/src/zlib/libzlib.a OpenDB/src/tm/libtm.a OpenDB/src/defin/libdefin.a /Users/ronaldv/Projects/repositories/msaligane/OpenROAD-flow/tools/OpenROAD/src/OpenDB/src/def/lib/libdef.a OpenDB/src/defout/libdefout.a OpenDB/src/lefout/liblefout.a -ltcl OpenDB/src/lefin/liblefin.a OpenDB/src/zutil/libzutil.a /Users/ronaldv/Projects/repositories/msaligane/OpenROAD-flow/tools/OpenROAD/src/OpenDB/src/lef/lib/liblef.a FastRoute/third_party/fastroute/libFastRoute4.1.a flute3/libflute.a FastRoute/third_party/pdrev/libpdrev.a -ldl OpenPhySyn/src/StandardTransforms/BufferFanoutTransform/libbuffer_fanout.a OpenPhySyn/src/StandardTransforms/GateCloningTransform/libgate_clone.a OpenPhySyn/src/StandardTransforms/PinSwapTransform/libpin_swap.a OpenPhySyn/src/StandardTransforms/ConstantPropagationTransform/libconstant_propagation.a /Users/ronaldv/Projects/repositories/msaligane/OpenROAD-flow/tools/OpenROAD/src/OpenSTA/app/libOpenSTA.a

jjcherry56, 2020-07-17:
I develop openroad on macos. I don't use the tcl in macos because it is lacking beyond supporting tclsh. Use homebrew to install tcl-tk and it should work without any command line options.

As for CUDD, see src/OpenSTA/README.md. It is optional and if you are having problems, don't use it.

closed 2020-07-17 by jjcherry56

20Mhz, 2020-07-17:
Hi James, 
I'm also able to build on macOS, the request is to have an option on the build script to properly pass the variables  to the CMAKE scripts, currently I'm  forced to do that manually. Right now it seems just easier to skip the build script altogether and build manually.
Bests,
Ronald

rovinski, 2020-07-17:
@20Mhz If your issue is with OpenROAD-flow, please file the issue under that repository.

20Mhz, 2020-07-18:
Ah! True, didn't realize it's the wrong repository. Thanks