The-OpenROAD-Project/OpenROAD #386: TritonCTS support for SDC

kentropy, 2020-07-18:
There is a SDC support limitation in Yosys synthesis, but no obvious comment about this on tritonCTS is documented. Is there any other source of reference that I can check out to get further information?

rovinski, 2020-07-18:
@MarceloDanigno any documentation on TritonCTS limitations? I remember there used to be something, but I can't find it.

@kentropy TritonCTS supports quite a bit of the general CTS use cases. It might be easier to ask if the specific features you are looking for are supported.

maliberty, 2020-07-18:
All of the openroad app relies on OpenSTA to read SDC and provides access
to the result.  Generally OpenSTA has very good SDC support.  Is there
something particular you are concerned about?  You can see more information
at
https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/doc/OpenSTA.pdf

TritonCTS mostly uses the SDC to find clock roots afaik.

Matt

MarceloDanigno, 2020-07-23:
Sorry about the PR above, it is unrelated to this issue (mean to tag 389).
The limitations to TritonCTS that I found are issues regarding the clock tree builder algorithm. For example, having a fanout factor higher than 2. Most of the others were already close either from updates to TritonCTS (multiple clocks, characterization) or OpenSTA (finding clock roots).
All SDC information in TritonCTS is received from OpenSTA, and we currently only use it to find clock roots, as Matt said.

rovinski, 2020-07-23:
@MarceloDanigno I think the question is more about the restrictions on what types of sdc-defined clocks TritonCTS supports. e.g. generated clocks, unbalanced waveforms, negative-edge flops, etc. I think that since @kentropy hasn't responded this issue can be closed.

closed 2020-07-23 by rovinski

kentropy, 2020-07-23:
Sorry for late replies. What I mean is more towards the modeling parameters that TritonCTS supports. E.g.: max skew, max fanout, ..., and so on. 
I think the document @maliberty mentioned is what I was looking for. Thank you all!