The-OpenROAD-Project/OpenROAD #397: Sink/source verification at the end of a global placment seems to fail on nets connected to pins

rbarzic, 2020-07-23:
When running a global placement on my design, I get an error message (when fastroute is invoked apparently):

Initializing nets...
Checking pin placement...
Checking pin placement... Done!
 > ----Checking sinks/source...
[ERROR] Net CLKIN has invalid sinks/source distribution
    #Sinks: 0; #sources: 1
[ERROR] Net DBG0 has invalid sinks/source distribution
    #Sinks: 0; #sources: 0
[ERROR] Net DBG1 has invalid sinks/source distributio

....

A testcase is available at https://github.com/rbarzic/issues-openroad. directory replace1. It should be self contained
I'm using Openroad, branch openroad  sha1 811dc2ff

rbarzic, 2020-07-23:
Apparently it seems to be a regression as I was able to try the same example with an openroad version from April  (sha1 : 9183af4f) and I don't have error (It looks like some extra checks were added after this date)

rovinski, 2020-07-23:
Might be related to #395 (@MarceloDanigno ?)

rbarzic, 2020-07-23:
I don't think this is related- #395 seems to be about TritonCTS. Here there is a check that has been added to check for partially connected nets but it seems to not take in account the case of a net connected to the pad window of an  I/O cell - Net will have only one connection to an instance in this case

eder-matheus, 2020-07-23:
@rbarzic Could you move this issue to FastRoute repository? It seems to be a wrong verification in FastRoute

rbarzic, 2020-07-23:
@eder-matheus - Apparently, according to https://docs.github.com/en/github/managing-your-work-on-github/transferring-an-issue-to-another-repository, I may not have the right to transfert the issue myself