The-OpenROAD-Project/OpenROAD #408: The OpenROAD-Flow repository is closed

kevingo, 2020-07-28:
Hi, why the [OpenROAD-flow](https://github.com/The-OpenROAD-Project/OpenROAD-flow) repo was closed ?

rovinski, 2020-07-31:
We are aware of the issue and working to resolve it as quickly as possible.

kevingo, 2020-08-01:
@rovinski Thanks.

rashmi2383, 2020-08-14:
Do we have an ETA on this?

rovinski, 2020-08-15:
@rashmi2383 unfortunately no ETA at this time. This issue will be closed once it is back up. You can put a watch on this issue to be notified by GitHub when that happens.

rovinski, 2020-08-18:
A temporary read-only copy is now up at https://github.com/The-OpenROAD-Project/OpenROAD-flow

We are still working on getting the full history restored.

rashmi2383, 2020-08-19:
I am seeing some build issues on Ubuntu 18.04 with OpenROAD-flow. Can I open a new issue in this repo?

rovinski, 2020-08-19:
Is it #482 ?

rashmi2383, 2020-08-19:
This is the error:

[INFO][FLOW-0000] using local build method. This will create binaries at tools/build/
[100%] Building yosys
[100%] Building abc/abc-7a98ef0
[100%] Building yosys-filterlib
[100%] Building share/xilinx/brams_init_36.vh
[100%] Building share/xilinx/brams_init_32.vh
[100%] Building share/xilinx/brams_init_18.vh
[100%] Building share/xilinx/brams_init_16.vh
/usr/bin/ld: cannot find -ltcl8.5
[100%] Building share/ice40/brams_init1.vh
[100%] Building share/ice40/brams_init2.vh
collect2: error: ld returned 1 exit status
passes/techmap/Makefile.inc:66: recipe for target 'yosys-filterlib' failed
make: *** [yosys-filterlib] Error 1
make: *** Waiting for unfinished jobs....
ERROR: ABC contains local modifications! Set ABCREV=default in Yosys Makefile!
Makefile:622: recipe for target 'abc/abc-7a98ef0' failed
make: *** [abc/abc-7a98ef0] Error 1
/usr/bin/ld: cannot find -ltcl8.5
collect2: error: ld returned 1 exit status
Makefile:573: recipe for target 'yosys' failed
make: *** [yosys] Error 1
make: *** wait: No child processes.  Stop.

rovinski, 2020-08-19:
It's looking for tcl8.5 because CentOS 7 is the official platform. If you have tcl8.6 then just replace it in the build script:
```
sed -i 's/tcl8\.5/tcl8\.6/' build_openroad.sh
```
If you don't have tcl installed, then install it:
```
apt-get install tcl-dev
```
If this doesn't work for you then please open a new issue.

rashmi2383, 2020-08-19:
This helped. But I still see the Yosys Makefile error:

ERROR: ABC contains local modifications! Set ABCREV=default in Yosys Makefile!
Makefile:622: recipe for target 'abc/abc-7a98ef0' failed
make: *** [abc/abc-7a98ef0] Error 1
make: *** Waiting for unfinished jobs....

rovinski, 2020-08-19:
@rashmi2383 please open a new issue, this is not related to the thread.

tspyrou, 2020-10-15:
This is the new area for the flow. https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts

closed 2020-10-15 by tspyrou