The-OpenROAD-Project/OpenROAD #413: Could not find a package configuration file provided by "Eigen3"

jahagirdar, 2020-08-03:
While compiling the code I faced the following error:
```
-- STA executable: /prj/openroad/OpenROAD/src/OpenSTA/app/sta
CMake Error at src/replace/CMakeLists.txt:25 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

After Digging around on stackoverflow, I understand that this has happened due to changes between CMake 2.x and Cmake 3.x.

My system configuration is:
cmake --version
cmake version 3.16.3

lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 20
Release:	20
Codename:	ulyana

rovinski, 2020-08-03:
The error means what it says. `eigen3` is a dependency. You need to install it on your system.

closed 2020-08-03 by rovinski

maliberty, 2020-08-03:
You can look at the Docker file to see the dependencies, e.g.:

# eigen required by replace, TritonMacroPlace
RUN git clone https://gitlab.com/libeigen/eigen.git \
    && cd eigen \
    && mkdir build \
    && cd build \
    && cmake .. \
    && make install