The-OpenROAD-Project/OpenROAD #416: PDNGEN Core Ring Connection Problem

viyelesay, 2020-08-04:
In my design, I use Metal1 for rails, Metal 6 for straps. Core ring consists of Metal 5 and Metal 6. Since Metal 5 is not used in rails or straps, pdngen gives error that it could not find Metal5. It does not also increase the length of Metal1 to connect on Metal6 on the ring. It only connects Metal1 to Metal6 on the strap.

rovinski, 2020-08-04:
@viyelesay please include screen shots and a copy of the log file (sanitizing any proprietary data as necessary). If you can reproduce the issue using open source data (e.g. Nangate45 / Skywater130) it would help significantly.

viyelesay, 2020-08-05:
Please check the attached file.
[pdngenissue.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/5026220/pdngenissue.zip)

Colin-Holehouse, 2020-08-11:
Thanks for the clear testcase, I was able to make some modifications to the code to allow for the metal5/metal6 core rings and the metal1/metal6 grid over the stdcells - you are correct in thinking that you will need to have the {metal5 metal6} connection defined in order to connect the core rings properly.

Extending the stdcell rail out to the core ring, wasn't something I had been looking to add, but based on your request I will add an option in the config file to make that happen, something like:

rails {
  metal1 {width 0.17 pitch 2.4 offset 0 extend_to_core_ring 1}
}

I've checked the code updates into the branch pdn-issue_416 if you want to try it out. Please let me know if you have any further comments - the branch will need to go through a few more QA steps before it is merged in with the the openroad branch, I will update here as and when the merge happens.

viyelesay, 2020-08-12:
Thanks for the quick response and fix. I checked your solution and it is working. Is it also possible to select number via rows between rails and core connections? Currently, there are only one via row connection. It will be nice if there is a parameter that we control number of via rows.

Colin-Holehouse, 2020-08-12:
Currently connections to the core rings are made in exactly the same way as to the power straps. The code looks to create vias that fit within the overlapping areas of the layers involved. If there was space for more via rows then they should be added (I would regard this as a bug if it did not happen like this). 

It's not clear to me why you think it would be better to have multiple rows of vias for the connections to the core rings, but if I understand your comment correctly, you're asking for a larger via to be inserted that is wider than the metal1 rail. Given that we're talking about a via stack from metal1 up to metal6, then we are talking about several different vias up through the stack - would you want control over each individual via in the stack, or the same number of rows to be used for all vias in the stack? 

Constraints can be added for connections - there are not many of these defined currently, e.g. max_columns which limits the number of columns that are present in a via. Perhaps we can add a new constraint here "min_columns" / "min_rows" potentially that would allow you to force the via to be larger than the minimum size.

viyelesay, 2020-08-12:
Connections to the core ring are made the same way as the power straps. There is no bug in terms of that.

The main reason to add more via between core ring and metal1 straps is to reduce the via resistance on the power lines. Since there is enough space on the core rings, we can add couple of extra via rows.

Controlling each individual via stack will be nice, but I don't think it is necessary.

closed 2021-04-20 by maliberty