The-OpenROAD-Project/OpenROAD #417: TritonCTS optimization subject to given constraints

kentropy, 2020-08-05:
In the original paper of TritonCTS, it stated that the algorithm performs optimizations subject to the given constraints like maximum transition time, load capacitance, and fanout.

I did a little experiment by changing the constraint of max fanout in SDC file with a fixed placement. But the results seems to be the same. I expect TritonCTS to keep optimize until the constraint is met.

mathursanjiv, 2021-02-24:
Normally clock tree fanout is different from data path max fanout. This is set in TritonCTS using -sink_clustering_size option. This is typically used with -sink_clustering_max_diameter to keep cluster load reasonable. In absence of -sink_clustering_size and -sink_clustering_max_diameter, clustering size would be determined by load cap of clusters.
Clock tree building uses characterization to ensure Max cap and max transition violation do not occur.

maliberty, 2023-10-17:
quite old with no further discussion

closed 2023-10-17 by maliberty