The-OpenROAD-Project/OpenROAD #430: Detail placer optimization to stay on lower metals

quentinw91, 2020-08-07:
Hi,
I modified the resize.tcl script to have buffers placed at each input pin + antenna cells.  (the script just searches by names the input buffers and then adds the antenna cells, similar to what is done in antenna_fix.tcl).  

````
...
# Set the buffer cell
set buffer_cell [get_lib_cell [lindex $::env(MIN_BUF_CELL_AND_PORTS) 0]]
set_dont_use $::env(DONT_USE_CELLS)

puts "Perform input port buffering..."
buffer_ports -inputs -buffer_cell $buffer_cell

puts "Adding antenna cells to inputs buffers...."
#antenna cell name is hard-coded for now, TODO: set up a env variable
set antenna_cell_name "ANTENNA_X1"
set db [::ord::get_db]
set block [[$db getChip] getBlock]
set antenna_master [$db findMaster $antenna_cell_name]
set input_antenna_cells_count 0
foreach inst [$block getInsts] {
    #puts "instance name: [$inst getName]"
    if { [regexp {^input[0-9]+$} [$inst getName] whole number] } {
        puts "matched name: $whole"
        set buffer_iterm [$inst findITerm "A"]
        set input_net [$buffer_iterm getNet]
        puts "net name: [$input_net getName]"

        set antenna_inst_name "ANTENNA"
        append antenna_inst_name "_" [$input_net getName]

        set inst_loc_x [lindex [$inst getLocation] 0]
        set inst_loc_y [lindex [$inst getLocation] 1]
        set inst_ori [$inst getOrient]

        set antenna_inst [odb::dbInst_create $block $antenna_master $antenna_inst_name]
        set antenna_iterm [$antenna_inst findITerm "A"]

        $antenna_inst setLocation $inst_loc_x $inst_loc_y
        $antenna_inst setOrient $inst_ori
        $antenna_inst setPlacementStatus PLACED
        odb::dbITerm_connect $antenna_iterm $input_net

        set input_antenna_cells_count [expr $input_antenna_cells_count + 1]
        puts "Added [$antenna_inst getName] to [$input_net getName] net."
    }
}
puts "Added [expr $input_antenna_cells_count] antenna cells."

puts "Perform buffer insertion..."
set_max_fanout $::env(MAX_FANOUT) [current_design]
repair_design -max_wire_length $::env(MAX_WIRE_LENGTH) -buffer_cell $buffer_cell
...
````
I tested this with gcd example (nangate45).  It works nice, but i am facing an issue with the detail placement afterwards (see screenshots attached).  The placer places the antenna cells on the "wrong" side of the buffer:
A: This is not optimum (the router could connect both terminals with METAL1 if it was placed on the right side, instead of having to go in METAL3)
B: In this case, it even causes a "functional" issue.  If the wire from the pin is very long (in METAL2), it could cause an antenna error because there is a METAL3 connection between the antenna diode and the gate to protect...

Is there a way I could hint the placer how to place this ? (I thought of using the orientation to modify the location of the antenna cell wrt the buffer, but there is no notion of the instance orientation at the resize level yet...)
Thank you

full script: [resize.tcl.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/5040410/resize.tcl.txt)

[image]
[image]

jjcherry56, 2020-08-19:
the hint would be to place the diode on the side you want it to end up on.
if you put the diode in an empty site (not simple) the detailed placer won't move it

quentinw91, 2020-08-19:
The problem is that i do not know yet on which side it should go. The orientation of the buffer is only decided at the placement step ( i am guessing to try to reduce congestion).  So i will only know the desired position after placement...
I was wondering if there is a way of putting a different weight for a specific net (and depending on metal "heigth").  That way, when the placer decides on the orientation of the buffer (or the localization of the antenna cell wrt the buffer), the optimization will lead it to keep this net at lower metals.

jjcherry56, 2020-08-19:
The orientation won't change unless you call optimize_mirroring, so don't do that.
No weights used in detailed placement.

closed 2021-02-15 by jjcherry56