The-OpenROAD-Project/OpenROAD #450: Pin Offgrid in ioPlacer

viyelesay, 2020-08-16:
In my process, Met2 min width is 0.28, min Met2 area is 0.202 and manufacturing grid is 0.005. ioPlacer calculates the height of pin as 0.722, which is close to 0.202/0.28. However, since the manufacturing grid is not 0.001, drc gives offgrid error. Is it possible to ceil the height of the pin to integer multiple of manufacturing grid and also set  coordinates of the pin on the manufacturing grid?

maliberty, 2020-08-16:
Can you provide a simple test case?

viyelesay, 2020-08-16:
Hi, I created a patch which sets pin height according to manufacturing grid. Once the height is set, positions are also on the grid.
[ioPlacerOffGridPatch.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/5080851/ioPlacerOffGridPatch.zip)

viyelesay, 2020-08-16:
Please check the attached file. I have to add area property for metal2 and metal3 in the technology lef file to show the issue.
[offgridissue.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/5080924/offgridissue.zip).

viyelesay, 2020-08-24:
The patch that I sent was for old version. Please change tech with _tech in the patch before you apply it.

juspertor, 2021-01-01:
I have the same issue here. The problem is in line 613 and 638 of IOPlacer.cpp. The shape size is calculated by:

    int height = int(std::max(2.0 * halfWidth, ceil(_core.getMinAreaX() / (2.0 * halfWidth))));

without any check of the manufacturinggrid. In case the layer minimum area is set in the TLEF file the resulting height may result in a value that is not on the manufacturinggrid. This results in a DRC violation of the design. The height value should be round up to the next manufacturinggrid value. 

From the IOPlacer.cpp source code it seems that the vertical-length / horizontal-length can be set by a parameter. But after a first review, I did not find a way to set this parameter from OpenROAD. The required procs are not forwarded to the OpenROAD interface. Is there a way to set this value?

eder-matheus, 2021-04-08:
The fix for the issue is already available at the public repo. Can it be close?

closed 2021-04-14 by eder-matheus