The-OpenROAD-Project/OpenROAD #455: Router is assuming all the pins to be in the first layer

mukul19147, 2020-08-17:
Even if I supply pins in the different layer the router is assuming them in the first layer and routing because of which many higher-layer pins remain unconnected. How to fix it?

vijayank88, 2023-06-14:
@eder-matheus 
Is required any further response?

eder-matheus, 2023-06-14:
@vijayank88 @mukul19147 I would only add an observation regarding the `set_routing_layers` command, where you can set specific layer ranges for signal and clock nets. But note that even with the layer range defined, the global router needs to create guides to the pin layers for pin access. So you may see routing out of the layer range only for pin access.

maliberty, 2023-10-17:
@arthurjolo your current work should resolve this

closed 2023-10-26 by maliberty