The-OpenROAD-Project/OpenROAD #457: OpenROAD is crashing while starting FastRoute

tamimcse, 2020-08-17:
OpenROAD is crashing while starting FastRoute for https://github.com/tamimcse/test/blob/master/top.v. The Verilog is generated using Bambu HLS. The OpenROAD script is https://github.com/tamimcse/test/blob/master/bash_script.sh. The output is as following: You will see the crash at the very end.

`.....
......
yosys> opt_clean -purge

Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module _Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy..
Removed 96 unused cells and 9494 unused wires.
<suppressed ~183 debug messages>
y_typical.lib iberty /opt/panda/share/panda//nangate45/lib/NangateOpenCellLibrary

Printing statistics.
=== _Z4sailPhPtS_PjS_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S1_S_S_S_yy ===

Number of wires: 20066
Number of wire bits: 21196
Number of public wires: 2469
Number of public wire bits: 3599
Number of memories: 0
Number of memory bits: 0
Number of processes: 0
Number of cells: 18580
AND2_X1 1925
AND2_X2 49
AND2_X4 45
AND3_X1 488
AND3_X2 3
AND3_X4 6
AND4_X1 120
AND4_X2 3
AND4_X4 3
AOI211_X1 272
AOI211_X2 3
AOI21_X1 882
AOI21_X2 3
AOI221_X1 1
AOI221_X2 2
AOI221_X4 33
AOI22_X1 22
BUF_X1 4070
BUF_X16 2
BUF_X2 115
BUF_X4 273
BUF_X8 5
CLKBUF_X2 58
CLKBUF_X3 3
DFF_X1 1514
INV_X1 807
INV_X2 18
INV_X32 5
INV_X4 6
LOGIC0_X1 1
MUX2_X1 1143
MUX2_X2 4
NAND2_X1 1288
NAND2_X2 3
NAND2_X4 1
NAND3_X1 451
NAND3_X2 1
NAND4_X1 84
NAND4_X2 1
NOR2_X1 1262
NOR2_X2 11
NOR2_X4 9
NOR3_X1 335
NOR3_X2 1
NOR3_X4 1
NOR4_X1 58
NOR4_X2 2
[... 445 lines cut ...]
fixed cells : 1716
nets : 22618
design area : 1438127936000.000
total fixed area : 1825824000.000
total movable area : 287720496000.000
design utilization : 20.032
rows : 428
row height : 2800
Check Legality
row check ==> PASS
site check ==> PASS
power check ==> PASS
edge_check ==> PASS
placed_check ==>> PASS
overlap_check ==> PASS
-------------------- Placement Analysis ------------------------
total displacement : 1657681
average displacement : 76
max displacement : 16853
original HPWL : 763228.133
legalized HPWL : 763580.620
delta HPWL : 0%
Adjust layer 2 in 70.0%
Adjust layer 3 in 70.0%
*** buffer overflow detected ***: openroad terminated`

maliberty, 2021-02-24:
@eder-matheus is this resolved?

vijayank88, 2023-06-14:
Required repo link missing to reproduce the issue.
Try with latest OpenROAD flow, please re-open with test case again, if there is crash still.

closed 2023-06-14 by vijayank88