The-OpenROAD-Project/OpenROAD #463: What is the meaning of the parameter in IP_global.cfg？

ChibiMarukoZ, 2020-08-17:
I have a problem with the floorplan phase of OpenRoad-flow. 

Its macro positioning is always wrong and it reports segment fault. 

Finally I found **IP_global.cfg**, but I don't understand what the parameter inside means

> // Global Fin 
> set ::FIN_PITCH 0.0025
> // Global Row Height
> set ::ROW_HEIGHT 1.4
> // Global Placment Height
> set ::SITE_WIDTH 0.19
> // Global Halo Width Vertical Edge
> set ::HALO_WIDTH_V 15.12
> // Global Halo Width Horizontal Edge
> set ::HALO_WIDTH_H 22.4
> //Global Vertical Channel Width
> set ::CHANNEL_WIDTH_V 19.95
> //Global Horizontal Channel Width
> set ::CHANNEL_WIDTH_H 18.8

ヽ(✿ﾟ▽ﾟ)ノ Hope to get help form you.

jjcherry56, 2020-08-19:
look in src/TritonMacroPlace/doc/

ChibiMarukoZ, 2020-11-05:
Thank you for solving the problem ！

closed 2021-04-29 by ZhishengZeng