The-OpenROAD-Project/OpenROAD #468: Segfault in test-1-large-pin-count test-case

oharboe, 2020-08-17:
To reproduce:

Get this pull request for alpha-release: https://github.com/The-OpenROAD-Project/alpha-release/pull/28

Then run:

```
# cd alpha-release/flow/
# make DESIGN_CONFIG=designs/test-1-large-pin-count.mk
[deleted]
INFO:  Now linking: LargePinCount
INFO:  Successfully linked: LargePinCount
NO PATH !
INFO: Total 0 worst path!
/bin/bash: line 17:    69 Segmentation fault      (core dumped) RePlAce -bmflag etc -lef ./objects/nangate45/LargePinCount/merged_padded.lef -def ./results/nangate45/LargePinCount/2_floorplan.def -verilog ./results/nangate45/LargePinCount/2_floorplan.v -lib ./platforms/nangate45/NangateOpenCellLibrary_typical.lib -sdc ./results/nangate45/LargePinCount/2_floorplan.sdc -output ./results/nangate45/LargePinCount/replace -t 1 -timing -resPerMicron 1.59 -capPerMicron 0.235146e-12 -skipIP -plot -experi output -den 0.509 -initCoef 0.00002 -onlyGP 2>&1
        70 Done                    | tee ./logs/nangate45/LargePinCount/3_1_RePlAce.log

```

maliberty, 2020-08-18:
Is there anything to be fixed in this issue?  It isn't clear to me.

oharboe, 2020-08-18:
@maliberty  Run the test on the latest version of OpenROAD to see if the problem is still there.

maliberty, 2021-11-28:
Having heard nothing since Aug I'm closing this issue.

closed 2021-11-28 by maliberty