The-OpenROAD-Project/OpenROAD #478: Prevent Resizer from resizing specific cells

quentinw91, 2020-08-18:
Hi,
When adding buffer cells to the input pins (using ``` buffer_ports -inputs ``` command), they get resized by the resizer script.  Even tough i specified MIN_BUF_CELL_AND_PORTS = BUFX1, some of the input pins buffers get resized to BUFX2 or BUFX4.  
This can be seen also on gcd example with the following resize.tcl script , where the BUFX1 input buffers get resized to CLK_BUFX1's
```
...
# Set the buffer cell
set buffer_cell [get_lib_cell [lindex $::env(MIN_BUF_CELL_AND_PORTS) 0]]
set_dont_use $::env(DONT_USE_CELLS)

puts "Perform input port buffering..."
buffer_ports -inputs -buffer_cell $buffer_cell

set db [::ord::get_db]
set block [[$db getChip] getBlock]
foreach inst [$block getInsts] {
    #puts "instance name: [$inst getName]"
    if { [regexp {^input[0-9]+$} [$inst getName] whole number] } {
        puts "matched name: $whole"
        set inst_master [$inst getMaster]
        put "master is : [$inst_master getName]"
        #i tried marking the input buffer as FIXED, to avoid being resized by resizer, but it did not work
 #       $inst setPlacementStatus FIRM
    }
}

puts "Perform buffer insertion..."
set_max_fanout $::env(MAX_FANOUT) [current_design]
repair_design -max_wire_length $::env(MAX_WIRE_LENGTH) -buffer_cell $buffer_cell

# Perform resizing
puts "Perform resizing after buffer insertion..."
resize
...
```

Is there a way to prevent Resizer to resize specific cells ? (like marking them as DONT_RESIZE or something)
What i am hoping for is that in case of slew rate repair, Resizer adds a new buffer after the existing one, instead of replacing it (BUFX1-->BUFX4--> (rest of logic)..., instead of BUFX4-->(rest of logic)...) if it is marked as DONT_RESIZE.  
My goal here is to not modify the capacitance presented at input pin, but i guess that other people could have situations where they do not want resizer to resize specific cells.

Is there something allowing this in Resize code ?
Thanks

jjcherry56, 2020-08-19:
why not just swap the command order?

quentinw91, 2020-08-19:
Well then i was wondering how would Resizer resize the logic behind the input buffer without knowing what is driving it, but i guess that if i specify  in the sdc a ````set_driving_cell ```` for each input pin corresponding to the input buffer, it would know...
I will try this, Thanks !

jjcherry56, 2020-08-19:
set_driving_cell

quentinw91, 2020-08-27:
Hi,
apparently the ```set_driving_cell``` won't solve this because it will get in the way of Resizer to repair max_fanout violation (see #496 ).  So i think I should try this other approach to solve my issue : (without the need to modify Resizer for my specific case)

I use the ```buffer_ports -input``` and let Resizer resize them if needed (that way the max_fanout is solved). 
And then search for all buffer inputs that were resized, and "re-buffer" them.
For that, I would need to have a function to put an input buffer to a specific pin.  Is it possible to add it ?

I guess the code should look like that:

Lines to add to Resize.tcl:
```
proc parse_pin { keys_var required } {
  upvar 1 $keys_var keys
  set pin "NULL"
  if { [info exists keys(-pin)] } {
    set pin_name $keys(-pin)
    if { $pin_name != "" } {
      set pin [get_pin_error "-pin" $pin_name]
      if { $pin != "NULL" } {
	if { ![get_property $pin is_pin] } {
	  ord::error "[get_name $pin] is not a pin."
	}
      }
    }
  } elseif { $required } {
    ord::error "-pin required for pin buffer insertion."
  }
  if { $pin == "NULL" && $required } {
    ord::error "-pin required for pin buffer insertion."    
  }
  return $pin
}

define_cmd_args "buffer_input_port" {-pin pin\
				  -buffer_cell buffer_cell}

proc buffer_input_port { args } {
  parse_key_args "buffer_input_port" args \
    keys {-buffer_cell -pin} flags {}

  set buffer_cell [parse_buffer_cell keys 1]
  set pin [parse_pin keys 1]
  check_argc_eq0 "buffer_input_port" $args

  buffer_input_port $pin $buffer_cell
}
```

and lines to add to Resize.i:
```
void
buffer_input_port(Pin *input_pin,
			   LibertyCell *buffer_cell)
{
  ensureLinked();
  Resizer *resizer = getResizer();
  resizer->bufferInput(input_pin, buffer_cell);
}
```
I am not sure this is the only files that should be modified.  And I am a bit confused about what is a "pin" and a "port", which seems to be a bit mixed in the code (especially for the ```is_pin``` verification in ```parse_pin``` proc...).  So could you maybe review this and add it if it is okay ?

Thanks

vijayank88, 2023-06-14:
@quentinw91 
Still are you facing the issue?
Provide right test case to reproduce the same.

maliberty, 2024-09-03:
stale

closed 2024-09-03 by maliberty