The-OpenROAD-Project/OpenROAD #482: [OpenROAD-flow-public] yosys seems to not build correctly

quentinw91, 2020-08-19:
(i would have put this issue in the OpenROAD-flow-public repo, but it is not possible to open issues there because it has been archived)
Hi,
Trying to build the OpenROAD-flow-public on Ubuntu 18.04.4, i do not see errors in the log file:
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/5096622/build_openroad.log)

But i get an error saying yosys is not found:
````
bash-4.2$ source ./setup_env.sh 
OPENROAD: /OpenROAD-flow/tools/OpenROAD
bash-4.2$ cd flow/
bash-4.2$ make synth
[INFO][FLOW] Using platform directory ./platforms/nangate45
/bin/bash: yosys: command not found
make: *** [versions.txt] Error 127
````
env seems be to set correctly
````
TERM=xterm
OLDPWD=/OpenROAD-flow
OPENROAD=/OpenROAD-flow/tools/OpenROAD
PATH=/OpenROAD-flow/tools/build/OpenROAD/src:/OpenROAD-flow/tools/build/TritonRoute:/OpenROAD-flow/tools/build/yosys/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
PWD=/OpenROAD-flow/flow
SHLVL=1
HOME=/
_=/usr/bin/env
````
but yosys src directory does not contain bin:
````
bash-4.2$ ls -lsa /OpenROAD-flow/tools/build/yosys/
total 22796
    4 drwxr-xr-x 2 root root     4096 Aug 19 12:29 .
    4 drwxr-xr-x 1 root root     4096 Aug 19 12:29 ..
 7628 -rwxr-xr-x 1 root root  7808848 Aug 19 10:46 yosys
15068 -rwxr-xr-x 1 root root 15429400 Aug 19 12:15 yosys-abc
    4 -rwxr-xr-x 1 root root     3222 Aug 19 12:15 yosys-config
   32 -rwxr-xr-x 1 root root    31320 Aug 19 12:15 yosys-filterlib
   56 -rwxr-xr-x 1 root root    53737 Aug 19 12:15 yosys-smtbmc
````

rovinski, 2020-08-19:
It's just a bad submodule hash.

Run the build script first, then do:
```
cd tools/yosys
git checkout master
make install -j$(nproc) PREFIX=../build/yosys CONFIG=gcc TCL_VERSION=tcl8.5
```
I'll see if we can get the hash fixed.

tspyrou, 2020-08-19:
fixed on the openroad branch.

rovinski, 2020-08-19:
Not actually fixed yet, workaround still required.

rovinski, 2020-08-19:
Now it's fixed.

closed 2020-08-19 by rovinski

quentinw91, 2020-08-20:
It does not work for me...  I checked out the repo (master branch) and reran a full build.  log file: 
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/5102346/build_openroad.log)
still have yosys missing
i tried to run the ````make install...```` also, did not change

rovinski, 2020-08-20:
If you build using docker then the tools get installed in a docker image, not the local file system (per the README). Install & run in Docker works on my side (although `qt5-qtbase-devel` is missing as a dependency in the Dockerfile).

If you have Docker installed but want a local build, use `build_openroad.sh --local`. Local builds work for me on CentOS 6, 7 and Ubuntu 18.04.

maliberty, 2020-08-20:
> (although `qt5-qtbase-devel` is missing as a dependency in the Dockerfile).

yosys does not depend on qt5; openroad does and it is in the Dockerfile.

rovinski, 2020-08-21:
Yes, I'm just pointing out that someone will run into that issue if they try to use the OpenROAD-flow Docker install.

quentinw91, 2020-08-21:
I know how to build the project with docker, i am still doing it with the old repo configuration (that i checked out somewhere in July before the crash) and that works fine...
I was just pointing that the new repo (OpenRoad-flow-public) does not work for me: yosys seems to be incorrectly build on the docker image (there is not "bin" or "share" directories in "/OpenROAD-flow/tools/build/yosys/", only:
```
bash-4.2$ ls -lsa /OpenROAD-flow/tools/build/yosys/
total 22796
    4 drwxr-xr-x 2 root root     4096 Aug 19 12:29 .
    4 drwxr-xr-x 1 root root     4096 Aug 19 12:29 ..
 7628 -rwxr-xr-x 1 root root  7808848 Aug 19 10:46 yosys
15068 -rwxr-xr-x 1 root root 15429400 Aug 19 12:15 yosys-abc
    4 -rwxr-xr-x 1 root root     3222 Aug 19 12:15 yosys-config
   32 -rwxr-xr-x 1 root root    31320 Aug 19 12:15 yosys-filterlib
   56 -rwxr-xr-x 1 root root    53737 Aug 19 12:15 yosys-smtbmc
```
As i do not see directly an error in the log file that could explain this, i am wondering what could cause this.  Was there a change in the dependencies from July to now ?