The-OpenROAD-Project/OpenROAD #483: [OpenROAD-flow-public] Error while building on Ubuntu 18.04

rashmi2383, 2020-08-19:
ERROR: ABC contains local modifications! Set ABCREV=default in Yosys Makefile!
Makefile:622: recipe for target 'abc/abc-7a98ef0' failed
make: *** [abc/abc-7a98ef0] Error 1
make: *** Waiting for unfinished jobs....

rovinski, 2020-08-19:
Try cleaning yosys and rebuilding (using the workaround in #482).

rashmi2383, 2020-08-19:
This error doesn't go away.

rovinski, 2020-08-19:
Try a fresh clone.

rashmi2383, 2020-08-19:
Tried but no luck. Rest everything passes except this.

rovinski, 2020-08-19:
I don't know what else to tell you, this issue is not reproducible for me on CentOS 6/7 or Ubuntu 18.04. The issue is self-explanatory, your ABC directory is not clean. I don't know what's happening to not make it not clean, but that shouldn't happen if you simply do:
```
git clone https://github.com/The-OpenROAD-Project/OpenROAD-flow OpenROAD-flow
cd OpenROAD-flow
./build_openroad.sh
```

rashmi2383, 2020-08-19:
Using ./build_openroad.sh without --local switch helped.

closed 2020-08-19 by rovinski