The-OpenROAD-Project/OpenROAD #496: Resizer does not take driving_cell max fanout into account

quentinw91, 2020-08-26:
Hi,
After what was discussed in #478, i am using the ```buffer_ports -inputs``` command after resize, to prevent it from resizing the input buffers.  I am using ```set_driving_cell``` to hint resizer what will drive these nets, but it does not look to consider the driving_cell max_fanout.
To reproduce it, i modified the nangate45 .lib file with ```default_max_fanout  : 20;``` statement and provided a netlist with an input pin (RESETN) driving more than 20 cells:
[testcase_drivingcell.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/5128676/testcase_drivingcell.zip)

The input buffers are correctly added, but i get the following violation:

```
...
==========================================================================
pin_count
--------------------------------------------------------------------------
110

Perform buffer insertion...
Perform resizing after buffer insertion...
Resized 0 instances.
Perform input port buffering...
Inserted 2 input buffers.
Repair tie lo fanout...
Repair tie hi fanout...
Repair hold violations...
No hold violations found.

==========================================================================
report_floating_nets
--------------------------------------------------------------------------

==========================================================================
report_checks
--------------------------------------------------------------------------
Startpoint: _25_ (rising edge-triggered flip-flop clocked by CLK)
Endpoint: _26_ (rising edge-triggered flip-flop clocked by CLK)
Path Group: CLK
Path Type: max

    Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------
           0.00    0.00    0.00   clock CLK (rise edge)
                   0.00    0.00   clock network delay (ideal)
           0.00    0.00    0.00 ^ _25_/CK (DFFR_X1)
   2.06    0.01    0.10    0.10 ^ _25_/Q (DFFR_X1)
           0.01    0.00    0.10 ^ _26_/D (DFFR_X1)
                           0.10   data arrival time

           0.00    2.00    2.00   clock CLK (rise edge)
                   0.00    2.00   clock network delay (ideal)
                   0.00    2.00   clock reconvergence pessimism
                           2.00 ^ _26_/CK (DFFR_X1)
                  -0.03    1.97   library setup time
                           1.97   data required time
-----------------------------------------------------------------------
                           1.97   data required time
                          -0.10   data arrival time
-----------------------------------------------------------------------
                           1.87   slack (MET)

Startpoint: _39_ (rising edge-triggered flip-flop clocked by CLK)
Endpoint: _40_ (rising edge-triggered flip-flop clocked by CLK)
Path Group: CLK
Path Type: min

    Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------
           0.00    0.00    0.00   clock CLK (rise edge)
                   0.00    0.00   clock network delay (ideal)
           0.00    0.00    0.00 ^ _39_/CK (DFFR_X1)
   1.12    0.01    0.08    0.08 v _39_/Q (DFFR_X1)
           0.01    0.00    0.08 v _40_/D (DFFR_X1)
                           0.08   data arrival time

           0.00    0.00    0.00   clock CLK (rise edge)
                   0.00    0.00   clock network delay (ideal)
                   0.00    0.00   clock reconvergence pessimism
                           0.00 ^ _40_/CK (DFFR_X1)
                   0.00    0.00   library hold time
                           0.00   data required time
-----------------------------------------------------------------------
                           0.00   data required time
                          -0.08   data arrival time
-----------------------------------------------------------------------
                           0.08   slack (MET)

==========================================================================
report_tns
--------------------------------------------------------------------------
tns 0.00

==========================================================================
report_wns
--------------------------------------------------------------------------
wns 0.00

==========================================================================
report_slew_violations
--------------------------------------------------------------------------
max fanout

Pin                                   Limit Fanout  Slack
---------------------------------------------------------
input2/Z                                 20     22     -2 (VIOLATED)

==========================================================================
report_design_area
--------------------------------------------------------------------------
Design area 149 u^2 2% utilization.

==========================================================================
instance_count
--------------------------------------------------------------------------
138

==========================================================================
pin_count
--------------------------------------------------------------------------
114
```

Would it be possible to have resizer consider the driving_cell max_fanout ?  Or maybe have an additional parameter to OpenSTA ```set_driving_cell``` function to let the user specify the max_fanout ? (it could be interesting if the user does not specify a driving cell, but only a input_transition time for a given max fanout)
Thanks

rovinski, 2020-08-26:
`set_max_fanout` in the SDC is typically the way you globally constrain fanout.

`repair_design` is the command that fixes DRVs, including fanout. You have to call that after `buffer_ports`.

quentinw91, 2020-08-27:
```repair_design``` does not consider input ports with the specified ```set_max_fanout```command.   I tried it with the testcase above:
```
...
set_max_fanout $::env(MAX_FANOUT) [current_design]
repair_design -max_wire_length $::env(MAX_WIRE_LENGTH) -buffer_cell $buffer_cell
...
```
If i do not use ```buffer_ports```, i get more than 20 cells on RESETN net, and it is not reported as a max_fanout violation.  
I do not know why it does not consider input ports (and if it is done on purpose or not).  Buf if it is on purpose, having a ```set_driving_cell``` with a max_fanout option would allow the user to specify a max_fanout on input pins.

(And no i do not want to use ```repair_design``` after ```buffer_ports```, because it will resize the input buffer (see #478 ) )