The-OpenROAD-Project/OpenROAD #497: set_power_net global variable

viyelesay, 2020-08-26:
power and ground variables in set_power_net function in opendp.tcl needs to be defined as global. Otherwise, they are destroyed once the function ends. Therefore, detailed_placement gives an error that it could not find the power nets.

rovinski, 2020-08-26:
I don't follow what the problem is. `opendp.tcl` is an interface file to translate userspace commands into internal commands. `set_power_net` doesn't set userspace variables, it alters database objects.

Please attach a test case showing the faulty behavior.

viyelesay, 2020-08-27:
in opendp.tcl "opendp::set_power_net_name $power" is executed. $power is a string and is passed to "Opendp::setPowerNetName(const char *power_name)" in the Opendp.cpp file. However, power_name in Opendp.cpp is defined as a pointer, therefore the string must reside in the memory after "proc set_power_net { args } {" procedure ends. To see the faulty behaviour I printed power_net_name_ variable in the "Opendp::findRowPower()" in dbToOpendp.cpp file.

rovinski, 2020-08-27:
I think that description is sufficient, however it is significantly easier for our dev team if issues have a test case. i.e.:
1. The inputs to reproduce the problem (e.g. a script)
2. The output log which demonstrates the problem
3. Info about your system such as OS version and software version (commit hash)

Easier for dev team = faster bug fixing = better for everyone 🙂

rovinski, 2020-08-27:
@jjcherry56

viyelesay, 2020-08-27:
Thanks for your response. My OS is Centos 7.

jjcherry56, 2020-08-27:
I removed the command so it isn't an issue. You won't see the change until the public repo is updated.
3122b623 rm set_power_net cmd

closed 2020-08-27 by jjcherry56