The-OpenROAD-Project/OpenROAD #499: SWIG version inconsistency

ax3ghazy, 2020-08-31:
The README states that SWIG 4.0 is needed:
https://github.com/The-OpenROAD-Project/OpenROAD/blame/openroad/README.md#L16

However, the cmake lists in a couple of occasions require 3.0:

https://github.com/The-OpenROAD-Project/OpenDB/blob/9ccc0bfb7cd7ec70892a7cdacce113303bfa7f7f/CMakeLists.txt#L41

https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/CMakeLists.txt#L294

I am guessing that the intention was to allow 3.0 as the minimum version? The `REQUIRED` cmake option will only match 3.x versions. The only way, as far as I understand, was to do something like the following, which was omitted 7 months ago:

https://github.com/The-OpenROAD-Project/OpenDB/commit/d9c69262f60d347f97b3966b8d4b61334638520c

Thanks.

maliberty, 2020-09-15:
I have a machine with only swig 4.0 and can build openroad without any problem.  find_packages isn't limited to 3.0 unless the keyword EXACT is used, which it isn't.  Have you observed a failure?  If so please show what it is.

ax3ghazy, 2020-09-15:
On further inspection, the issue is that 

```
find_package(SWIG REQUIRED 3.0)
```
should be
```
find_package(SWIG 3.0 REQUIRED)
```

What it's currently trying to do is find a component "3.0" from SWIG (e.g., `find_package(Qt COMPONENTS QtCore)`), which doesn't exist, so it fails with the following error (which kinda looks like a error about an incorrect version):
```
CMake Error at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find SWIG (missing: 3.0) (found version "4.0.2")
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.18/Modules/FindSWIG.cmake:105 (find_package_handle_standard_args)
  src/CMakeLists.txt:86 (find_package)
```

(https://cmake.org/cmake/help/latest/command/find_package.html#id2)

maliberty, 2020-09-15:
That makes more sense.  We can fix that.

ax3ghazy, 2020-09-23:
This has been resolved; thank you. Closing this.

closed 2020-09-23 by ax3ghazy