The-OpenROAD-Project/OpenROAD #500: [OpenROAD-flow-public] Error while making gcd design in the flow directory

rashmi2383, 2020-09-03:
I have OpenROAD-flow installed on centos. I was trying to build the gcd design flow and I am seeing the following error:

[INFO][FLOW] Using platform directory ./platforms/nangate45
(/usr/bin/time -f "%Eelapsed %PCPU %MmemKB" stdbuf -o L klayout -zz -rd design_name=gcd \
        -rd in_def=./results/nangate45/gcd/6_final.def \
        -rd in_gds="./platforms/nangate45/gds/NangateOpenCellLibrary.gds  " \
        -rd seal_gds="" \
        -rd out_gds=results/nangate45/gcd/6_1_merged.gds \
        -rd tech_file=./objects/nangate45/gcd/klayout.lyt \
        -rm ./util/def2gds.py) 2>&1 | tee ./logs/nangate45/gcd/6_1_merge.log
stdbuf: failed to run command 'klayout': No such file or directory
0:00.00elapsed 100%CPU 696memKB
make: *** [results/nangate45/gcd/6_1_merged.gds] Error 127

Can someone please guide on what is going wrong here?

rovinski, 2020-09-03:
You need to install [KLayout](https://www.klayout.de/).

closed 2020-09-21 by jjcherry56

rashmi2383, 2020-09-22:
After I fixed the klayout issue, now i see this error:

[INFO][FLOW] Using platform directory ./platforms/nangate45
mkdir -p ./results/nangate45/gcd ./logs/nangate45/gcd ./reports/nangate45/gcd
(/usr/bin/time -f "%Eelapsed %PCPU %MmemKB" yosys -c scripts/synth.tcl) 2>&1 | tee ./logs/nangate45/gcd/1_1_yosys.log
/usr/bin/time: cannot run yosys: No such file or directory
0:00.00elapsed 68%CPU 368memKB
make: *** [results/nangate45/gcd/1_1_yosys.v] Error 127

maliberty, 2020-09-22:
Did yosys compile correctly?  Is it on your path?

rashmi2383, 2020-09-22:
Yes, it did. I will check the path though.

rashmi2383, 2020-09-22:
That helped. Thanks!