The-OpenROAD-Project/OpenROAD #502: FR antenna avoidance flow

agorararmard, 2020-09-09:
I'm trying out the antenna avoidance flow in the latest FR in the OpenROAD app. I tried the test case on the repo and it worked fine. I tried another DEF/LEF from the sky130 hd library (after adding ANTENNADIFFAREA  0.434700 ; in the diode cell LEF) and I got this error:
```
Old total capacity: 15850
[WARNING]No OR_DEFAULT vias defined
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
Notice 0: Split top of 1 T shapes.
[INFO] #Antenna violations: 13
Warning: could not find power special net
Error: S 
```
I tried out multiple test cases and this symbol after Error changes with every run so it's probably a memory corruption issue and it should be a segmentation fault. I tried different utilizations as well, down to 10% and up to 65%. So, if you can help point me to what's wrong, what I'm doing wrong, or display a more meaningful error message. Attached is one of the test cases. 

[FR.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5196364/FR.tar.gz)

agorararmard, 2020-09-11:
I noticed that a lot of updates were added to FR yesterday. So, I tried out the newest version and the behavior of the test case above (after modifying the script to the new commands) changed but it was still faulty.

So, attached here are two test cases using the same script (a variant from the [test script under FR](https://github.com/The-OpenROAD-Project/OpenROAD/blob/openroad/src/FastRoute/test/diode_insert.tcl)).
- spm_invalid_guides: A relatively small design that passes Fastroute, but:
1. The antenna violations are not detected.
2. The guides are invalid. They don't carry any layer information.

- xtea_segfault: A slightly larger design that segfaults during the antenna avoidance flow process after detecting 100 antenna violations.
[FR_antenna.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5210892/FR_antenna.tar.gz)

agorararmard, 2020-09-23:
@eder-matheus 

Once again, I tried out the latest OR, and the first case above is resolved (spm_invalid_guides). 

However, the second case is not yet resolved. It now detects 99 antenna violations and then segfaults. So I'm attaching a renewed version of the test case. The script used is a variant from the test script you provide [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/openroad/src/FastRoute/test/repair_antennas1.tcl).

Please, let me know if I'm doing something wrong or if there is a bug causing this.

[FR_testcase_xtea.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5269857/FR_testcase_xtea.tar.gz)

maliberty, 2020-09-23:
I've fixed the error handling bug so that now it prints the proper message from the detailed placer:

Error: utilization exceeds 100%.

eder-matheus, 2020-09-23:
@agorararmard You're running the repair_antennas after filler insertion. If you use the DEF before filler insertion, you will not see this error again.
I'm not sure if FastRoute should remove the fillers before repair_antennas, or if we should change the flow to run filler insertion after FastRoute, but at the moment you will need to do this workaround.

maliberty, 2020-09-23:
@eder-matheus @rovinski repair_antennas is not part of the current flow but when it is we will need to move filler insertion as you suggest to after repair.

agorararmard, 2020-09-24:
@eder-matheus moving fill_insertion to after global_routing worked. But, the generated def file and guides didn't pass through TR14. I'm still going to investigate it now. but here is the error TR14 showed:
```
post process guides ...                                                                                                                       
GCELLGRID X -1 DO 77 STEP 6900 ;                                                                                                              
GCELLGRID Y -1 DO 76 STEP 6900 ;                                                                                                              
  complete 10000 orig guides                                                                                                                  
  complete 20000 orig guides                                                                                                                  
  complete FR_MASTERSLICE                                                                                                                     
  complete FR_VIA                                                                                                                             
  complete li1                                                                                                                                
  complete mcon
  complete met1
  complete via
  complete met2
  complete via2
  complete met3
  complete via3
  complete met4
  complete via4
  complete met5
Error: genGuides_split lineIdx is empty on met2
```
I'm still going to look into it, but, in case you wanted to take a look, attached are the guides produced by FastRoute, and the DEF after running global routing/diode insertion and fill insertion on it. Have you tried running the output successfully through TR14 before?

Also, I noticed that the overlap test check fails in check_placement. This also happens in the [test case](https://github.com/The-OpenROAD-Project/OpenROAD/blob/openroad/src/FastRoute/test/repair_antennas1.ok) you have in the repo. So, I'm wondering what are the implications of that.

[TR_input.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5278699/TR_input.tar.gz)

agorararmard, 2020-09-25:
@eder-matheus, writing the def after repair_antenna and re-running FastRoute, then rewriting the def and guides solved the issue (produced valid guides and a def file with diodes), which is a workaround/hack. So, my flow now looks roughly like this:
```
FastRoute
FR::repair_antennas
FastRoute
fill_insertion
TR14
```

The overlap failure is not causing any DRC violations, so I'm guessing I could ignore it. 

Also, it would be nice to print the number of diodes inserted somewhere in the log or in a separate report. because I noticed that the number of diodes added != antenna violations found.

eder-matheus, 2020-09-25:
@agorararmard Could you send me your Tcl scripts to run FastRoute? The guides generated after repair_antennas command should be correct for the DEF with diodes.
The reported overlap is probably a false positive, due to the way the diodes are inserted. But I'll double-check it.
I will improve the report for repair_antennas in my next PRs, thanks for the feedback.

agorararmard, 2020-09-25:
@eder-matheus here is the tcl script I'm using in the flow: https://github.com/efabless/openlane/blob/develop-FR_antenna/scripts/openroad/or_route.tcl

agorararmard, 2020-09-29:
@eder-matheus, I have a separate question. Why does using `repair_antenna` require reading a liberty file, while ARC itself doesn't require reading a liberty file (for antenna checking)?

maliberty, 2020-09-29:
I have also pointed out this liberty issue and I believe @eder-matheus is already fixing it.

msaligane, 2020-09-29:
@agorararmard @maliberty I think he might have added reading .lib because of the diodes. But I do not know if the repair flow is aware of the timing impact due to additional load of inserted diodes.

@agorararmard 

> because I noticed that the number of diodes added != antenna violations found.

This is because sometimes you may need multiple diodes to clear one violation.

maliberty, 2020-09-29:
It is not timing aware and the usage of timing was just incidental and unnecessary.

agorararmard, 2020-10-15:
I've checked the latest openroad HEAD, and this has been solved. Thanks for your help!

However, the script still requires reading the liberty, but since it's unrelated to this issue, I'll close it.

agorararmard, 2020-10-20:
@eder-matheus, @maliberty: So, the issue appeared again in larger designs. Fastroute completes without complaining about anything, but then TritonRoute has an issue with the guides.
```
post process guides ...                                                                                                                       
GCELLGRID X -1 DO 77 STEP 6900 ;                                                                                                              
GCELLGRID Y -1 DO 76 STEP 6900 ;                                                                                                              
  complete 10000 orig guides                                                                                                                  
  complete 20000 orig guides                                                                                                                  
  complete FR_MASTERSLICE                                                                                                                     
  complete FR_VIA                                                                                                                             
  complete li1                                                                                                                                
  complete mcon
  complete met1
  complete via
  complete met2
  complete via2
  complete met3
  complete via3
  complete met4
  complete via4
  complete met5
Error: genGuides_split lineIdx is empty on met3
```
Here is the fastroute input
[fastroute_input.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5409487/fastroute_input.tar.gz)
Here is the TritonRoute input:
[tritonroute_input.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/5409491/tritonroute_input.tar.gz)

It's important to note that when doing:
```
FastRoute
FR::repair_antenna
FastRoute
Fill insertion
TritonRoute
```
this issue doesn't happen, which indicates that this is propably an issue with the guides of the rerouted nets in the repair antenna step.

eder-matheus, 2020-11-07:
@agorararmard I've updated openroad branch with the fix for this issue. TritonRoute should not fail anymore after repair_antennas command

closed 2021-01-06 by agorararmard